riscv/riscv-unified-db #322: "anyOf" is used where "allOf" should be

ThinkOpenly, 2024-11-27:
For some instructions, for the set of extensions in which they are "definedBy", there are instances where "anyOf" is used where I believe "allOf" is the correct condition. For example, for `fli.h`:
```
definedBy:
  anyOf: [Zfa, Zfh]
```
However, the unpriv spec (20240411) defines all of the "FLI" instructions in the "Zfa" extension, but for "FLI.H":
> If the Zfh or Zvfh extension is implemented, FLI.H performs the analogous operation [...]

So, "FLI.H" is only implemented if both "Zfa" _and_ "Zfh" are supported.

I think all of these should be "allOf" instead of "anyOf":
```
./arch/inst/Zfh/fli.h.yaml:  anyOf: [Zfa, Zfh]
./arch/inst/Zfh/fcvt.h.s.yaml:  anyOf: [Zfh, Zfhmin]
./arch/inst/Zfh/fmv.x.h.yaml:  anyOf: [Zfh, Zfhmin]
./arch/inst/Zfh/flh.yaml:  anyOf: [Zfh, Zfhmin]
./arch/inst/Zfh/fcvt.h.d.yaml:  anyOf: [D, Zfh]
./arch/inst/Zfh/fmaxm.h.yaml:  anyOf: [Zfa, Zfh]
./arch/inst/Zfh/fleq.h.yaml:  anyOf: [Zfa, Zfh]
./arch/inst/Zfh/fminm.h.yaml:  anyOf: [Zfa, Zfh]
./arch/inst/Zfh/froundnx.h.yaml:  anyOf: [Zfa, Zfh]
./arch/inst/Zfh/fltq.h.yaml:  anyOf: [Zfa, Zfh]
./arch/inst/Zfh/fsh.yaml:  anyOf: [Zfh, Zfhmin]
./arch/inst/Zfh/fcvt.s.h.yaml:  anyOf: [Zfh, Zfhmin]
./arch/inst/Zfh/fround.h.yaml:  anyOf: [Zfa, Zfh]
./arch/inst/Zfh/fcvt.d.h.yaml:  anyOf: [D, Zfh]
./arch/inst/D/fmvh.x.d.yaml:  anyOf: [D, Zfa]
./arch/inst/D/fcvtmod.w.d.yaml:  anyOf: [D, Zfa]
./arch/inst/D/fli.d.yaml:  anyOf: [D, Zfa]
./arch/inst/D/froundnx.d.yaml:  anyOf: [D, Zfa]
./arch/inst/D/fleq.d.yaml:  anyOf: [D, Zfa]
./arch/inst/D/fround.d.yaml:  anyOf: [D, Zfa]
./arch/inst/D/fltq.d.yaml:  anyOf: [D, Zfa]
./arch/inst/D/fmaxm.d.yaml:  anyOf: [D, Zfa]
./arch/inst/D/fmvp.d.x.yaml:  anyOf: [D, Zfa]
./arch/inst/D/fminm.d.yaml:  anyOf: [D, Zfa]
./arch/inst/Q/fmaxm.q.yaml:  anyOf: [Q, Zfa]
./arch/inst/Q/fminm.q.yaml:  anyOf: [Q, Zfa]
./arch/inst/Q/froundnx.q.yaml:  anyOf: [Q, Zfa]
./arch/inst/Q/fmvp.q.x.yaml:  anyOf: [Q, Zfa]
./arch/inst/Q/fcvt.q.h.yaml:  anyOf: [Q, Zfh]
./arch/inst/Q/fround.q.yaml:  anyOf: [Q, Zfa]
./arch/inst/Q/fcvt.h.q.yaml:  anyOf: [Q, Zfh]
./arch/inst/Q/fmvh.x.q.yaml:  anyOf: [Q, Zfa]
./arch/inst/Q/fltq.q.yaml:  anyOf: [Q, Zfa]
./arch/inst/Q/fli.q.yaml:  anyOf: [Q, Zfa]
./arch/inst/Q/fleq.q.yaml:  anyOf: [Q, Zfa]
```

AFOliveira, 2024-12-03:
@ThinkOpenly Sorry for taking so long to get back on this, I completely missed it, I'll be taking a close look to this and answer you ASAP.

AFOliveira, 2024-12-03:
@ThinkOpenly Thank you for this catch, I see the mistake I did now. It was a wrong assumption based on riscv-opcodes. Do you prefer to PR this yourself or can I just go ahead and fix it?

ThinkOpenly, 2024-12-06:
I'm out of the office for a few weeks. If you'd like to submit a PR yourself, I would be grateful!

ThinkOpenly, 2025-01-22:
This was fixed by #340