riscv/riscv-unified-db #347: `value': wrong number of arguments (given 0, expected 1) (ArgumentError)

ShashankVM, 2024-12-07:
**Describe the bug**
When running regression for floating point add instruction, the regression fails with a "wrong number of arguments" signature.

**To Reproduce**
Steps to reproduce the behavior:
1. On https://github.com/ShashankVM/riscv-unified-db/tree/main
1. Run ` ./do gen:html_manual MANUAL_NAME=isa VERSIONS=all`

**Text output/Error Messages**
````
Extension 'Zihintntl' is not in the database
Extension 'Zimop' is not in the database
Extension 'Zicond' is not in the database
Extension 'Zawrs' is not in the database
Extension 'Zacas' is not in the database
Extension 'Zabha' is not in the database
Extension 'Ztso' is not in the database
Extension 'Zicbom', version '1.0.0' is not defined in the database
Extension 'Zicboz', version '1.0.0' is not defined in the database
Extension 'Zicbop', version '1.0.0' is not defined in the database
Extension 'Zfh' is not in the database
Extension 'Zfbfmin' is not in the database
Extension 'Zvfbfmin' is not in the database
Extension 'Zvfbfwma' is not in the database
Extension 'Zfa' is not in the database
Extension 'Zfinx' is not in the database
Extension 'Zdinx' is not in the database
Extension 'Zhinx' is not in the database
Extension 'C', version '2.0.0' is not defined in the database
Extension 'Zca' is not in the database
Extension 'Zcf' is not in the database
Extension 'Zcd' is not in the database
Extension 'Zcb' is not in the database
Extension 'Zcmp' is not in the database
Extension 'Zcmt' is not in the database
Extension 'Zbkb' is not in the database
Extension 'Zbkc' is not in the database
Extension 'Zbkx' is not in the database
Extension 'Zvl32b' is not in the database
Extension 'Zvl64b' is not in the database
Extension 'Zvl128b' is not in the database
Extension 'Zvl256b' is not in the database
Extension 'Zvl512b' is not in the database
Extension 'Zvl1024b' is not in the database
Extension 'Zve32x' is not in the database
Extension 'Zve32f' is not in the database
Extension 'Zve64x' is not in the database
Extension 'Zve64f' is not in the database
Extension 'Zve64d' is not in the database
Extension 'Zvfhmin' is not in the database
Extension 'Zvfh' is not in the database
Extension 'Zvknha' is not in the database
Extension 'Zvknhb' is not in the database
Extension 'Zbkb' is not in the database
Extension 'Zbkc' is not in the database
Extension 'Zbkx' is not in the database
Extension 'Zknd' is not in the database
Extension 'Zkne' is not in the database
Extension 'Zknh' is not in the database
Extension 'Zksed' is not in the database
Extension 'Zksh' is not in the database
Extension 'Zkr' is not in the database
Extension 'Zkn' is not in the database
Extension 'Zks' is not in the database
Extension 'Zk' is not in the database
Extension 'Zvbb' is not in the database
Extension 'Zvbc' is not in the database
Extension 'Zvkb' is not in the database
Extension 'Zvkg' is not in the database
Extension 'Zvkned' is not in the database
[... 125 lines cut ...]
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:281:in `each'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:281:in `execute'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:219:in `block in invoke_with_call_chain'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:199:in `synchronize'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:199:in `invoke_with_call_chain'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:188:in `invoke'
        from /home/shashank/riscv-unified-db/backends/manual/tasks.rake:412:in `block (4 levels) in <top (required)>'
        from /home/shashank/riscv-unified-db/backends/manual/tasks.rake:411:in `each'
        from /home/shashank/riscv-unified-db/backends/manual/tasks.rake:411:in `block (3 levels) in <top (required)>'
        from /home/shashank/riscv-unified-db/backends/manual/tasks.rake:400:in `each'
        from /home/shashank/riscv-unified-db/backends/manual/tasks.rake:400:in `block (2 levels) in <top (required)>'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:281:in `block in execute'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:281:in `each'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:281:in `execute'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:219:in `block in invoke_with_call_chain'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:199:in `synchronize'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:199:in `invoke_with_call_chain'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:188:in `invoke'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:188:in `invoke_task'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `block (2 levels) in top_level'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `each'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `block in top_level'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:147:in `run_with_threads'
        from /home/shashank/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:132:in `top_level'
        from -e:1:in `<main>'

dhower-qc, 2024-12-10:
```
diff --git a/lib/idl/ast.rb b/lib/idl/ast.rb
index e113b0ee..ec9f8774 100644
--- a/lib/idl/ast.rb
+++ b/lib/idl/ast.rb
@@ -1803,7 +1803,7 @@ module Idl
         var = symtab.get(lhs.text_value)
         value_result = value_try do
           v = rhs.value(symtab)
-          var.value = (lhs.value & ~0) | ((v & 1) << idx.value(symtab))
+          var.value = (lhs.value(symtab) & ~0) | ((v & 1) << idx.value(symtab))
         end
         value_else(value_result) do
           var.value = nil
```

@ShashankVM, please apply this in your PR and link this issue.

dhower-qc, 2024-12-11:
Fixed in #334

closed 2024-12-11 by dhower-qc