riscv/riscv-unified-db #362: AnyOf vs AllOf

AFOliveira, 2024-12-18:
**Describe the bug**
Some instructions have AnyOf and AllOf wrongly assigned. I'm trying to catch them all and submit a PR soon.
@BrianAnakPintar can you please address some examples of that?

BrianAnakPintar, 2024-12-18:
Examples:
### froundnx.h
UDB
```yaml
kind: instruction
name: froundnx.h
long_name: No synopsis available.
description: |
  No description available.
definedBy:
  anyOf: [Zfa, Zfh]
```
riscv-opcodes
```json
"froundnx_h": {
    "encoding": "010001000101-------------1010011",
    "variable_fields": [
        "rd",
        "rm",
        "rs1"
    ],
    "extension": [
        "rv_zfh_zfa"
    ],
```

### froundnx.q
UDB
```yaml
name: froundnx.h
long_name: No synopsis available.
description: |
  No description available.
definedBy:
  anyOf: [Zfa, Zfh]
```
riscv-opcodes
```json
"froundnx_q": {
    "encoding": "010001100101-------------1010011",
    "variable_fields": [
        "rd",
        "rm",
        "rs1"
    ],
    "extension": [
        "rv_q_zfa"
    ],
    "match": "0x46500053",
    "mask": "0xfff0007f"
},
```
### fmvp.q.x
UDB
```yaml
name: fmvp.q.x
long_name: No synopsis available.
description: |
  No description available.
definedBy:
  anyOf: [Q, Zfa]
```
riscv-opcodes
```json
 "fmvp_q_x": {
    "encoding": "1011011----------000-----1010011",
    "variable_fields": [
        "rd",
        "rs1",
        "rs2"
    ],
    "extension": [
        "rv64_q_zfa"
    ],
    "match": "0xb6000053",
    "mask": "0xfe00707f"
    },
```

Others include: 
`fround.q`, `fmvp.d.x` and more.

AFOliveira, 2024-12-19:
I was looking to this more thoroughly and I think #350 was supposed to fix this. I also had it fixed on #340 but I though it was already done and closed it. @dhower-qc was it that you maybe missed one commit or was it that our communication was wrong at the time?

dhower-qc, 2024-12-19:
I converted the string "[Za, Zb]" -> the object { anyOf: [Za, Zb] }. I didn't distinguish between cases that should have been allOf instead of anyOf.

AFOliveira, 2024-12-19:
Now that I look at it, I misexplained myself at that time, my bad :). I'll reopen it then.

closed 2025-03-18 by AFOliveira