riscv/riscv-unified-db #367: Pseudo-instructions labeled as instructions

AFOliveira, 2024-12-19:
**Describe the bug**
Some pseudo-instructions are labeled as instructions. 
I believe most of the following follow that pattern

DEBUG: Instruction 'lpad' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'rev8' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'zext.h' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'zip' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'unzip' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'sspopchk.x1' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'sspush.x5' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'sspush.x1' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'sspopchk.x5' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'gorci' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'xperm16' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'shfli' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'unshfli' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'xperm32' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'grevi' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'mop.rr.n' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'mop.r.n' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'amocas.d' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fleq.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fcvt.l.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fcvt.q.lu' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fli.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fminm.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fsgnjn.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fsub.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fcvt.d.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fmul.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fcvt.q.wu' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fsqrt.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fround.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fcvt.q.d' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fmax.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fcvt.wu.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fcvt.w.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'feq.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fsq' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fnmsub.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'flq' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fmadd.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fmsub.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fmvp.q.x' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fltq.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fdiv.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fcvt.q.l' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fcvt.s.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fcvt.q.h' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fmvh.x.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fsgnj.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fmin.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fnmadd.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fclass.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fcvt.lu.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fle.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'flt.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'froundnx.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fmaxm.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fcvt.h.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fsgnjx.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fcvt.q.s' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fcvt.q.w' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'fadd.q' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'c.sq' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'c.lq' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'c.sqsp' (from YAML) not found in JSON, skipping...
DEBUG: Instruction 'c.lqsp' (from YAML) not found in JSON, skipping...

AFOliveira, 2024-12-19:
@dhower-qc For now, can I just switch 

kind: instruction
to
kind: pseudoinstruction

and modify the schema to allow it, or should I have something else in mind?

dhower-qc, 2024-12-19:
Most (all?) of those FP instructions are real (e.g., fclass.q, fdiv.q, ...).

The quadword integer insts (c.lq, ...) would be real if RV128 was ratified, but it isn't (and it's not -- or at least shouldn't be -- included in UDB).

The bitmanip instructions (rev8, etc) are also real. I notice that it seems to be picking up the bitmanip instructions that are either only defined in RV32/64 or have different encodings in RV32/64. 

The CFI instructions (lpad, ss*) are pseudoinstructions (of the "new operation" type).

AFOliveira, 2024-12-19:
My bad, I didn't verify the full list and assumed that since the first ones were wrongly formatted - lpad, rev8, zext.h, zip, unzip- I assumed it was the same error for all, which is clearly not true. However, does my solution hold up?

AFOliveira, 2024-12-19:
I went through the files committed and I believe some were missed, should I reopen #340 , rebase and merge?

dhower-qc, 2025-03-28:
@AFOliveira, do we have this covered now?

AFOliveira, 2025-03-28:
> [@AFOliveira](https://github.com/AFOliveira), do we have this covered now?

I believe so, especially given that the instructions in the list are clearly instructions. I can't remember what exactly I was talking about, let's close it.

closed 2025-03-28 by AFOliveira