riscv/riscv-unified-db #369: $copy support was lost by PR #350

james-ball-qualcomm, 2024-12-20:
If you generate MC100 in the main branch, you'll see this problem. I also have a regression test for it (test_yaml_loader.rb tests) in my branch. 

[image]

Here's the test:

def test_copy_in_the_same_document
    yaml = <<~YAML
      $defs:
        target1: A string
        target2:
          a: hash
        target3: Another string

      obj1:
        target10: abc
        target11:
          $copy: "#/$defs/target1"
        target12: def
        target13:
          $copy: "#/$defs/target3"

    YAML

    doc = resolve_yaml(yaml)
    assert_equal({
        "$child_of" => "#/$defs",
        "target10"  => "abc",
        "target11"  => "A string",
        "target12"  => "def",
        "target13"  => "Another string"
      }, doc["obj1"])
  end

  # XXX
  def test_copy_in_the_different_document
    yaml2 = <<~YAML
      $defs:
        target1: A string
        target2:
          a: hash
        target3: Another string
    YAML

    yaml1 = <<~YAML
      obj1:
        target10: abc
        target11:
          $copy: "YAML2_REL_PATH#/$defs/target1"
        target12: def
        target13:
          $copy: "YAML2_REL_PATH#/$defs/target3"
    YAML

    doc = resolve_multi_yaml(yaml1, yaml2)
    assert_equal({
        "$child_of" => "test/test2.yaml#/$defs",
        "target10"  => "abc",
        "target11"  => "A string",
        "target12"  => "def",
        "target13"  => "Another string"
      }, doc["obj1"])
  end

dhower-qc, 2024-12-20:
Do we still need this? The one use case (MIE/MIP description) disappears if we go with community-driven prose + string-matched ratified text.

james-ball-qualcomm, 2024-12-20:
I'd like to wait to see how that Community Driven prose plus string match ratified text actually looks before we decide this isn't needed.

james-ball-qualcomm, 2024-12-20:
Also it was pretty easy to add support for this in the previous yaml loader code and the python version looks pretty similar so I assume adding it we'll also be easy. Also will probably start having more use cases that take advantage of $copy. The mip/mie was just the first I found.

dhower-qc, 2025-09-19:
I think not. One use case doesn't warrant a custom operator.

closed 2025-09-20 by ThinkOpenly