riscv/riscv-unified-db #388: Can't install python into devcontainer (SSL certificate failure)

james-ball-qualcomm, 2025-01-06:
I'm trying to run in a devcontainer under WSL2. 

```
root@46b914489016:/workspaces/riscv-unified-db# ./do RVA23
Using devcontainer environment
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:1000)'))': /simple/pre-commit/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:1000)'))': /simple/pre-commit/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:1000)'))': /simple/pre-commit/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:1000)'))': /simple/pre-commit/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:1000)'))': /simple/pre-commit/
Could not fetch URL https://pypi.org/simple/pre-commit/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pre-commit/ (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:1000)'))) - skipping
ERROR: Could not find a version that satisfies the requirement pre_commit==4.0.1 (from versions: none)
ERROR: No matching distribution found for pre_commit==4.0.1
UPDATE: Extracted base=32 from /workspaces/riscv-unified-db/arch/certificate_model/MC100-32.yaml
UPDATE: Extracted base=64 from /workspaces/riscv-unified-db/arch/certificate_model/MC100-64.yaml
UPDATE: Extracted base=32 from /workspaces/riscv-unified-db/arch/certificate_model/MC200-32.yaml
UPDATE: Extracted base=64 from /workspaces/riscv-unified-db/arch/certificate_model/MC200-64.yaml
UPDATE: Extracted base=32 from /workspaces/riscv-unified-db/arch/certificate_model/MC300-32.yaml
UPDATE: Extracted base=64 from /workspaces/riscv-unified-db/arch/certificate_model/MC300-64.yaml
UPDATE: Extracted base=64 from /workspaces/riscv-unified-db/arch/certificate_model/MockCertificateModel.yaml
UPDATE: Extracted base=64 from /workspaces/riscv-unified-db/arch/profile_release/MockProfileRelease.yaml
UPDATE: Extracted base=64 from /workspaces/riscv-unified-db/arch/profile_release/MockProfileRelease.yaml
UPDATE:   Calculated max_base=64 across profiles in MockProfileRelease
UPDATE: Extracted base=64 from /workspaces/riscv-unified-db/arch/profile_release/RVA20.yaml
UPDATE: Extracted base=64 from /workspaces/riscv-unified-db/arch/profile_release/RVA20.yaml
UPDATE:   Calculated max_base=64 across profiles in RVA20
UPDATE: Extracted base=64 from /workspaces/riscv-unified-db/arch/profile_release/RVA22.yaml
UPDATE: Extracted base=64 from /workspaces/riscv-unified-db/arch/profile_release/RVA22.yaml
UPDATE:   Calculated max_base=64 across profiles in RVA22
UPDATE: Extracted base=64 from /workspaces/riscv-unified-db/arch/profile_release/RVA23.yaml
UPDATE: Extracted base=64 from /workspaces/riscv-unified-db/arch/profile_release/RVA23.yaml
UPDATE:   Calculated max_base=64 across profiles in RVA23
UPDATE: Extracted base=64 from /workspaces/riscv-unified-db/arch/profile_release/RVB23.yaml
UPDATE: Extracted base=64 from /workspaces/riscv-unified-db/arch/profile_release/RVB23.yaml
UPDATE:   Calculated max_base=64 across profiles in RVB23
UPDATE: Extracted base=32 from /workspaces/riscv-unified-db/arch/profile_release/RVI20.yaml
UPDATE: Extracted base=64 from /workspaces/riscv-unified-db/arch/profile_release/RVI20.yaml
UPDATE:   Calculated max_base=64 across profiles in RVI20
UPDATE: Creating BaseArchitecture rv64 for /workspaces/riscv-unified-db/gen/profile_doc/adoc/RVA23.adoc
/workspaces/riscv-unified-db/.home/.venv/bin/python3 lib/yaml_resolver.py merge arch cfgs/rv64/arch_overlay gen/arch/rv64
Traceback (most recent call last):
  File "/workspaces/riscv-unified-db/lib/yaml_resolver.py", line 10, in <module>
    from tqdm import tqdm
ModuleNotFoundError: No module named 'tqdm'
/workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/file_utils.rb:65:in `block in create_shell_runner': Command failed with status (1): [/workspaces/riscv-unified-db/.home/.venv/bin/python3 lib/yaml_resolver.py merge arch cfgs/rv64/arch_overlay gen/arch/rv64] (RuntimeError)
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/file_utils.rb:57:in `sh'
        from /workspaces/riscv-unified-db/Rakefile:102:in `block in <top (required)>'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:281:in `block in execute'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:281:in `each'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:281:in `execute'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:219:in `block in invoke_with_call_chain'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:199:in `synchronize'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:199:in `invoke_with_call_chain'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:188:in `invoke'
        from /workspaces/riscv-unified-db/Rakefile:29:in `base_arch_for'
        from /workspaces/riscv-unified-db/backends/profile_doc/tasks.rake:60:in `block (2 levels) in <top (required)>'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:281:in `block in execute'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:281:in `each'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:281:in `execute'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:219:in `block in invoke_with_call_chain'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:199:in `synchronize'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:199:in `invoke_with_call_chain'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:243:in `block in invoke_prerequisites'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:241:in `each'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:241:in `invoke_prerequisites'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:218:in `block in invoke_with_call_chain'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:199:in `synchronize'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:199:in `invoke_with_call_chain'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:243:in `block in invoke_prerequisites'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:241:in `each'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:241:in `invoke_prerequisites'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:218:in `block in invoke_with_call_chain'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:199:in `synchronize'
[... cut ...]
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `each'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `block in top_level'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:147:in `run_with_threads'
        from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:132:in `top_level'
        from -e:1:in `<main>'
root@46b914489016:/workspaces/riscv-unified-db#

dhower-qc, 2025-01-06:
I think this is a Qualcomm IT issue; but we'll try to get a work-around in UDB

james-ball-qualcomm, 2025-01-06:
Derek and I did change Gemfile to use http instead of https and that solved the problem for Ruby gems but Python is a different beast.

[image]

closed 2025-02-22 by james-ball-qualcomm