riscv/riscv-unified-db #392: `jalr` IDL and description is unclear and potentially invalid

ThePuzzlemaker, 2025-01-07:
The ISA manual and Sail code specify that `jalr` clears the LSB of the calculated target address before jumping. The IDL code and description in this database do not specify this (not even in the `jump` function, to my knowledge), potentially making these entries invalid.

**Current value and location**
```
jump(X[rs1] + imm)
```
https://github.com/riscv-software-src/riscv-unified-db/blob/main/arch/inst/I/jalr.yaml#L30

**Expected value**
If I understand IDL correctly, it should be something vaguely like this:
```
jump({(X[rs1] + imm)[XLEN - 1:1], 1'b0})
```
The description should also be updated in my opinion to better clarify this behaviour.

ThinkOpenly, 2025-01-07:
Good catch!

You are correct about the `jump` function. It does not clear any bits. It will, however, raise an exception of the target address has low-order bits set (https://github.com/riscv-software-src/riscv-unified-db/blob/4d8b28a63a34be7262c0537962c0f3e8ef88bf43/arch/isa/globals.isa#L703):
```
    # raise a misaligned exception if address is not aligned to IALIGN
    if ((ialign() == 16) && ((target_addr & 0x1) != 0)) {
      # the target PC is not halfword-aligned
      raise(ExceptionCode::InstructionAddressMisaligned, mode(), target_addr);
    } else if ((target_addr & 0x3) != 0) {
      # the target PC is not word-aligned
      raise(ExceptionCode::InstructionAddressMisaligned, mode(), target_addr);
    }
```

Another approach with IDL (to my understanding of IDL):
```
jump((X[rs1] + imm) & ~1)
```

dhower-qc, 2025-01-08:
Yep, good catch. Either fix (Wren's or Paul's) should fix it. @ThePuzzlemaker, would you be able to make a PR for the fix?

PR #402 (2025-01-14): Clarify jalr description and IDL
Fixes #392
closed 2025-01-14 by dhower-qc