riscv/riscv-unified-db #407: Deployments Error

AFOliveira, 2025-01-10:
Deployments have been off for a few PRs now, For what I can see the problem was introduced on the refactor PR #350.

Probably everyone is already aware of it, but there isn't any issue targetting it.

[image]

dhower-qc, 2025-01-10:
Thanks, I hadn't noticed that. I'll fix it today.

kbroch-rivosinc, 2025-01-24:
Just adding some more context here since I'm looking at it:

"Generate YARDS task" rake task failing: https://github.com/riscv-software-src/riscv-unified-db/actions/runs/12381892807/job/34561480670#step:10:6

```
/home/runner/work/riscv-unified-db/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task_manager.rb:59:in `[]': Don't know how to build task '/home/runner/work/riscv-unified-db/riscv-unified-db/.stamps/dev_gems' (See the list of available tasks with `do --tasks`) (RuntimeError)
Did you mean?  /home/runner/work/riscv-unified-db/riscv-unified-db/.stamps
```

In the previous working run: https://github.com/riscv-software-src/riscv-unified-db/actions/runs/12381482637/job/34560141166#step:10:6. bundle is running to install the dev gems

```
bundle config set --local with development
bundle install
Updating files in /home/runner/work/riscv-unified-db/riscv-unified-db/.home/.cache
```

AFOliveira, 2025-01-28:
@dhower-qc I'm trying to fix this before Saturday, but I can't seem to run GH Actions outside the main branch due to branch protection, do you think you can configure a bypass for "pagesfix" branch so I can work on this? I believe that without it I can not test the GHA

dhower-qc, 2025-02-01:
fixed in #450

closed 2025-02-01 by dhower-qc