riscv/riscv-unified-db #408: add rubocop pre-commit hooks

kbroch-rivosinc, 2025-01-10:
As with the other code in the repo (python, yaml, json, etc), pre-commit hooks have been put in place to allow dev's to find and fix coding lint issue and consistently format the code before committing it.  Time for ruby?

Currently there is already a [.rubocop.yml](https://github.com/riscv-software-src/riscv-unified-db/blob/main/.rubocop.yml) file so I assume that's the starting point for what we want to lint/fmt.

If so, just need to add the [pre-commit cfg](https://github.com/rubocop/rubocop/blob/master/docs/modules/ROOT/pages/integration_with_other_tools.adoc#git-pre-commit-hook-integration-with-pre-commit)

Command line options of how rubocop runs in the pre-commit hook can be found here: https://github.com/rubocop/rubocop/blob/master/.pre-commit-hooks.yaml

@dhower-qc It is probably a bad idea to merge into main the autocorrected changes given how many other ruby related PRs are active but rather let the changes occur gradually as the files are edited for other reasons.  I am going to put 2 commits (one for pre-commit config, other with autocorrected changes) in the PR to show the autocorrected changes and let the CI run on them but I imagine removing the autocorrected changes before it is merged.  Thoughts?

Currently if I run rubocop manually:

```
❯ rubocop

...

47 files inspected, 2116 offenses detected, 1764 offenses autocorrectable

Tip: Based on detected gems, the following RuboCop extension libraries might be helpful:
  * rubocop-rake (https://rubygems.org/gems/rubocop-rake)

You can opt out of this message by adding the following to your config (see https://docs.rubocop.org/rubocop/extensions.html#extension-suggestions for more options):
  AllCops:
    SuggestExtensions: false
~/rvi/repos/riscv-software-src/riscv-unified-db on dev/kbroch/rubocop-pre-commit:main +1                      at 09:48:36 AM
```

Here's what's left when safe autocorrect (`-a`) was run previously:
```
❯ rubocop -a

...

47 files inspected, 411 offenses detected, 59 more offenses can be corrected with `rubocop -A`

Tip: Based on detected gems, the following RuboCop extension libraries might be helpful:
  * rubocop-rake (https://rubygems.org/gems/rubocop-rake)

You can opt out of this message by adding the following to your config (see https://docs.rubocop.org/rubocop/extensions.html#extension-suggestions for more options):
  AllCops:
    SuggestExtensions: false
~/rvi/repos/riscv-software-src/riscv-unified-db on dev/kbroch/rubocop-pre-commit:main +1 !38          took 4s at 09:58:15 AM
```

syedowaisalishah, 2025-03-18:
@ThinkOpenly , I have set up the RISC-V Unified Database in a Devcontainer and verified the Ruby environment.
I'd like to work on issue #408 and submit a PR. Can you assign it to me?

ThinkOpenly, 2025-03-18:
> I'd like to work on issue [#408](https://github.com/riscv-software-src/riscv-unified-db/issues/408)

@syedowaisalishah, I think this is already being worked in #409? Have you looked at that?

syedowaisalishah, 2025-03-18:
@ThinkOpenly Ok, any issue you suggest me to work on?