riscv/riscv-unified-db #410: Missing Zhinx Implementation

AFOliveira, 2025-01-10:
Zhinx extension has been added as needed by #340. However, only basic information has been added and needs completion.

batmnnn, 2025-01-30:
is this issue still open @AFOliveira ?

AFOliveira, 2025-01-30:
@batmnnn Yes, it is. I added the instructions from this extension but the rest of the data is missing.

AFOliveira, 2025-03-03:
@batmnnn Are you interested in contributing these?

batmnnn, 2025-03-03:
@AFOliveira i tried but was unable to fix it  , i'm a beginner so returned to basic open source repos.

AFOliveira, 2025-03-03:
If you want to try again and ask questions throughout the process, feel free to reach out! :)

dhower-qc, 2025-03-03:
> If you want to try again and ask questions throughout the process, feel free to reach out! :)

Definitely! We can help you through it.

batmnnn, 2025-03-03:
thanks , i'll surely try after my exams.

blazethunderstorm, 2025-07-16:
@AFOliveira @dhower-qc can i work on this pls