riscv/riscv-unified-db #412: H extension not included in RVA23

james-ball-qualcomm, 2025-01-12:
I've added the new profile-defined extension Sha and it implies a bunch of other profile-defined extensions as well as the H extension itself. However, I'm not seeing the H extension in the RVA23 and probably other similar profiles. I do see the profile-defined extensions implied by Sha.

[image]

dhower-qc, 2025-01-15:
How are you collecting the list of extensions?

james-ball-qualcomm, 2025-01-15:
I'll take a look.

dhower-qc, 2025-01-15:
Your understanding is correct -- `implies` means that if you specify B, you also get Zba, Zbc, and Zbs without having to explicitly specify them. `requires` means that if you specify D, you also must specify F or else you get a validation error.

Ruby checks `requires` in the validation checks.

You can get the full list, with `implies` expanded, using the `transitive*` functions of CfgArch

james-ball-qualcomm, 2025-01-15:
Okay, I've got a related question for you about the "implemented" functions of CfgArch. How do you actually use these "implemented" functions? I ask in particular because CfgArch only allows them to be called for a fully configured config. So I also don't allow these for portfolios since they correspond to a partially configured config. Should I allow them to be called for a portfolio? If yes, obviously I'll include all mandatory extensions but how should I treat the optional extensions? Should I add them too or omit them?