riscv/riscv-unified-db #428: Fence.tso is not pseudo for Fence

AFOliveira, 2025-01-20:
Currently UDB specified fence.tso as pseudo-instructions under [fence.yaml](https://github.com/riscv-software-src/riscv-unified-db/blob/d791ca4bb7c1938de313642f7dc0f04fde9d6769/arch/inst/I/fence.yaml#L206). However, this is not the case and fence.tso should have a separate file for it.

I believe this is the specified like this because of riscv-opcodes, but the only reason riscv-opcodes specified it like that is it not possible to support for both on current riscv-opcodes format.

Context:
https://github.com/riscv/riscv-opcodes/pull/330
https://github.com/riscv/riscv-isa-manual/issues/1782#issuecomment-2556036171

ThinkOpenly, 2025-01-21:
> fence.tso should have a separate file for it.

Probably so. It'll make a bit of a mess because you'll need to exclude the specific `FENCE.TSO` encoding from the `FENCE` encoding.

dhower-qc, 2025-01-21:
Agree. In terms of encoding, this is captured by the `hints` key (I might not have that in main yet; it's used by the ISS decoder). Maybe there is a better key name?

`fence.yaml`:

```yaml
# ...
hints:
  - { $ref: "inst/I/fence.tso.yaml" }
# ...
```

`fence.tso.yaml`:

```yaml
# full fence.tso instruction
```

AFOliveira, 2025-01-22:
I can't comprehend the hint part of fence.tso. Do they need to be connected at all? I think FENCE is described in the ISA both as a category of intructions (everything related to FENCEs) and as _just_ an instruction FENCE. 

[image]

I think it kind of acts the same way as the funct3/funct7 fields for Register-Register operations

[image]

 
 Can't we remove the fm field from fence as a variable field and just describe it as 0000 for FENCE and 0001 for FENCE.TSO or would that have any further implications?

ThinkOpenly, 2025-01-22:
How do we represent "reserved" values for which there is a "shall" in the ISA spec:
```
Likewise, many _fm_ [...] settings [...] are also reserved for future use. Base implementations shall treat all such reserved configurations as normal fences with fm=0000
```

> Can't we remove the fm field from fence as a variable field and just describe it as 0000 for FENCE and 0001 for FENCE.TSO or would that have any further implications?

That's actually what the Sail code does:
```
mapping clause encdec = FENCE(pred, succ)
  <-> 0b0000 @ pred @ succ @ 0b00000 @ 0b000 @ 0b00000 @ 0b0001111
[...]
mapping clause encdec = FENCE_TSO(pred, succ)
  <-> 0b1000 @ pred @ succ @ 0b00000 @ 0b000 @ 0b00000 @ 0b0001111
```

I note that you used `0001`, when the spec defines FENCE.TSO with _fm_=`0b1000`, but then I noticed that the IDL does the same (`0001`):
```
  if (fm == 1) {
    if (pred == 0x3 && succ == 0x3) {
      # this is a fence.tso instruction
      is_fence_tso = true;
    }
  }
```

Is the IDL code incorrect there? If we force FENCE to have _fm_=`0b0000`, then all the tests for values of _fm_ go away?

dhower-qc, 2025-02-11:
Yep, that's an IDL bug.

PR #460 (2025-04-29): fix(inst): make fence.tso a separate inst, fix fence bug
fixes #428
commit 01c02cd1 (2025-04-29): fix(inst): make fence.tso a separate inst, fix fence bug (#460)

fixes #428

---------
closed 2025-04-29 by dhower-qc