riscv/riscv-unified-db #431: Enhance IDL language & UDB tools to support linkable adoc anchors for Certification Test Plans

james-ball-qualcomm, 2025-01-21:
Want to be able to link to portions of IDL code in CTPs.

How about something like this? The anchor() statements would be part of the AST and the braces make it clear what they are related to. If we are going to tout IDL as the golden reference, we'll need some good way to link to portions of it. Does Sail have a solution for this?

```
operation(): |
  anchor("illegal-inst-exc-misa-disabled") {
    if (implemented?(ExtensionName::M) && (CSR[misa].M == 1'b0)) {
      raise (ExceptionCode::IllegalInstruction, mode(), $encoding);
    }
  }

  XReg src1 = X[rs1];
  XReg src2 = X[rs2];

  anchor("calculation") {
    X[rd] = (src1 * src2)[XLEN-1:0];
  }

Alasdair, 2025-01-24:
I hadn't considered referencing at a sub-function granularity, but if I pushed our documentation tooling's span feature down slightly further in the AST (a trivial change) you could do (roughly equivalent-ish Sail pseudocode):
```
function operation() = {
  $[span "illegal-inst-exc-misa-disabled"]
  if (implemented(Ext_M) & misa[M] == 0b0) {
    throw IllegalInstruction(mode(), encoding)
  }

  ...
}
```
Then you could reference from asciidoc using something like:
```
sail::illegal-inst-exc-misa-disabled[type=span]
```
Our documentation tooling currently only supports automatic cross-referencing from identifiers to top-level asciidoctor source blocks (i.e. you can click a function name and it'll take you to the asciidoc anchor for the block it's contained within). But if you wanted to reference a separate syntax-highlighted Sail file prepared with `sail --html` I think it would be fine.

james-ball-qualcomm, 2025-01-24:
Thanks Alasdair. How would the $[span ...] know the end of the cross reference? It's not just an anchor I'm looking for but a an anchor to a specific section of code. That's why I proposed the IDL syntax of anchor { } since the braces provide an anchor to the specified section of code. So for Sail, we'd some way to annotate where a "span" ends.

Alasdair, 2025-01-25:
Attributes in Sail apply to the subsequent expression/definition, which in that case would be the `if` expression (from the `if` keyword itself to the `}` two lines after). Blocks in Sail are expressions, so you could write
```
$[<attribute> <arguments>] { <expression> }
```
if you wanted to be explicit about the scope. The asciidoc tooling has various options for styling the contents so you can omit the braces or unindent the contents depending on what you need.

The syntax is very much inspired by Rust (which I think mostly copied C#), just with `$` rather than `#` and a more JSON-ish syntax for attribute arguments. https://doc.rust-lang.org/reference/attributes.html

james-ball-qualcomm, 2025-04-17:
# Add the following when https://github.com/riscv-software-src/riscv-unified-db/issues/431 is resolved.
  # [#idl:code:inst:mul:exception-illegal_instruction]#
  # if (implemented?(ExtensionName::M) && (CSR[misa].M == 1'b0)) {
  #   raise (ExceptionCode::IllegalInstruction, mode(), $encoding);
  # }#