riscv/riscv-unified-db #441: Commit schema checker doesn't seem to like a standard YAML array format

james-ball-qualcomm, 2025-01-25:
Commit schema checker can't support a standard YAML array format:

I get this error when I try to commit:

```
> git -c user.useConfigOnly=true commit --quiet --allow-empty-message --file -
bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Using singularity environment
check python ast.....................................(no files to check)Skipped
check builtin type constructor use...................(no files to check)Skipped
check for case conflicts.................................................Passed
check docstring is first.............................(no files to check)Skipped
check for broken symlinks............................(no files to check)Skipped
fix end of files.........................................................Passed
trim trailing whitespace.................................................Failed
- hook id: trailing-whitespace
- exit code: 1
- files were modified by this hook

Fixing arch/ext/Xmock.yaml

check for merge conflicts................................................Passed
check json...........................................(no files to check)Skipped
check yaml...............................................................Failed
- hook id: check-yaml
- exit code: 1

while scanning a plain scalar
  in "arch/ext/Xmock.yaml", line 167, column 13
found unexpected ':'
  in "arch/ext/Xmock.yaml", line 167, column 19
while scanning a plain scalar
  in "arch/csr/misa.yaml", line 144, column 17
found unexpected ':'
  in "arch/csr/misa.yaml", line 144, column 23
while scanning a plain scalar
  in "arch/inst/M/mul.yaml", line 77, column 13
found unexpected ':'
  in "arch/inst/M/mul.yaml", line 77, column 19
while scanning a plain scalar
  in "arch/inst/mock.yaml", line 68, column 13
found unexpected ':'
  in "arch/inst/mock.yaml", line 68, column 19

prettier.................................................................Passed
Validate files with jsonschema...........................................Passed
Validate files with jsonschema...........................................Passed
Validate files with jsonschema...........................................Passed
Validate files with jsonschema.......................(no files to check)Skipped
Validate files with jsonschema.......................(no files to check)Skipped
black................................................(no files to check)Skipped
pyupgrade............................................(no files to check)Skipped
```

I have to change my array from:
```
    links: [manual:inst:mul:encoding, udb:doc:inst:mock]
```

to 
```
    links: 
      - manual:inst:mul:encoding
      - udb:doc:inst:mock
```

to avoid the bogus error that thinks the array is a plain scalar even though the json schema for this yaml content is:

```
    "cert-coverage-points": {
      "description": "Architecturally visible behaviors requiring validation by certification tests",
      "type": "array",
      "required": ["id", "name", "links", "description"],
      "properties": {
        "id": {
          "type": "string"
        },
        "name": {
          "type": "string"
        },
        "links": {
          "description": "Cross reference to UDB documentation, ISA manual, or IDL code",
          "type": "array",
          "items": {
            "type": "string"
          }
        },
        "description": {
          "type": "string"
        }
      },
      "additionalProperties": false
    },
```

Nj221102, 2025-02-22:
Hi @james-ball-qualcomm, I will like to work on this issue, can you please assign it to me.

kbroch-rivosinc, 2025-03-18:
@james-ball-qualcomm : is this still an issue?  If so can you add the branch you are working on that has the problem?

dhower-qc, 2025-03-19:
`:` is a special character for YAML. You need to quote the strings, like so:

```yaml
 links: ["manual:inst:mul:encoding", "udb:doc:inst:mock"]
```