riscv/riscv-unified-db #445: Enhanced schema to represent conditional extension implications?

ThinkOpenly, 2025-01-27:
Some extensions don't fall neatly into using the existing "implies" and "requires" schema.

For example, the "C" extension has some somewhat complex implications (from ISA 20240411, chapter 27.2):
> The C extension is the superset of the following extensions:
> - Zca
> - Zcf if F is specified (RV32 only)
> - Zcd if D is specified
> 
> As C defines the same instructions as Zca, Zcf and Zcd, the rule is that:
> - C always implies Zca
> - C+F implies Zcf (RV32 only)
> - C+D implies Zcd

I don't think there is a way, currently, to represent the sort of conditional "implies" of the latter two rules.

Currently, the C extension doesn't imply anything, which is a bug, but based on the above:
- C should clearly imply Zca (we can implement this now, of course)
- C should imply Zcf _if_ F (RV32 only)
- C should imply Zcd _if_ D

I believe this would be required if instructions claimed to only be definedBy, say "Zca", but a configuration only defines "C"?

Similarly "Zce" (same page as above):
> Specifying Zce on RV32 without F includes Zca, Zcb, Zcmp, Zcmt
> Specifying Zce on RV32 with F includes Zca, Zcb, Zcmp, Zcmt and Zcf
> Specifying Zce on RV64 always includes Zca, Zcb, Zcmp, Zcmt

Oh! I see a TODO in `arch/ext/Zce.yaml` that describes this problem.

dhower-qc, 2025-01-29:
Thanks for pulling together this nice summary. I think we have the ability to specify these arbitrary dependencies for instruction definedBy, but not for implications/requirements. I *think* if we change the schema of implies/requires to match definedBy, then we can capture this.

dhower-qc, 2025-06-24:
Now handled.

closed 2025-06-24 by dhower-qc