riscv/riscv-unified-db #457: Put arch YAML files into their own repository

james-ball-qualcomm, 2025-02-05:
We've received feedback from the RVI community that it would be easier to contribute to the arch YAML files if they were in their own repository. It would be easier for these reasons:

1. Having the YAML files (i.e. the database) separated out from SW tools with make it easier to get their employer's permission to contribute.
2. There is a lower learning curve since the repository just has YAML files

dhower-qc, 2025-02-05:
I think we need a lot more information here. As discussed during the UDB meeting:

- The YAML files have executable code in them (IDL and Sail) and the arch/ directory contains non-YAML IDL files. As such, it's not clear to me how extracting arch/ helps.
- We need the code outside arch/ to run regressions on the data.
- Presumably, you also at least need the schemas/ directory

I'm not saying we can't reorganize, but we need to better understand why the current setup causes problems to know if the effort is worth it.

AFOliveira, 2025-02-27:
To update this discussion:

We may need to split UDB into  source-of-truth (arch/inst + cfgs)  on one repo and all tooling on another repo.
Moreover, there are a _lot_ of technical questions on how to do this - how do we ensure verification processes? do we keep all tooling to the side or do we keep a basic script like yaml_resolver.py to organize the DB?   

The way I see this is, if we split we will have:

Benefits - 
- Ease the adoption for those only insterestes in a structured source
- Ease people that may have trouble having company acceptance in participating in OSS work. Contributing to .yamls might be easier
- riscv-opcodes is mostly only instructions description - most software use is downstream - so to replace it, being as close to what it is, is valuable
- ratification process may also be eased by this, clearly we won't try to ratify the tooling part of UDB 

Cons - 
- Using submodules might add development "friction"
- Not all benefits are clear, there is a lot of uncertainty

james-ball-qualcomm, 2025-02-27:
> WARNING: This email originated from outside of Qualcomm. Please be wary of any links or attachments, and do not enable macros.

Afonso thanks for your comments and pros/cons. I've been imagining that the database repo contain everything in the arch directory (the yaml files), the associated schema files, and just enough software to check the yaml files against the schemas, and resolve the yaml files. My understanding is that this repo wouldn't require submodules.

Everything else (pretty much all the Ruby code) goes into a tool repository which at the database repo to its existing list of submodules.

What do people think of this partitioning?

+james+

AFOliveira, 2025-02-27:
> I've been imagining that the database repo contain everything in the arch directory (the yaml files), the associated schema files, and just enough software to check the yaml files against the schemas, and resolve the yaml files.

Just schema checking? We also need the other checks we are working, i.e. the LLVM one and riscv-opcodes verification. I think the CI may import this repo and perform that checks, but there is a gray area here.

ThinkOpenly, 2025-02-27:
In a former life, in dealing with lawyers and trying to get approval to work on open source projects where perhaps some of the project was considered "riskier" than other parts, we sometimes tailored proposals to restrict our participation to be entirely within certain subdirectories (and a promise to amend the proposal if the directories changed). For this project, if "all things Ruby, or Python, or other directly actionable programming language (thus excluding Sail and IDL)" have to meet a higher bar for getting permission to contribute to the project then perhaps we can ensure separation within the project between the data and the programs. Indeed, I believe we already do.

If we really need _some_ Python or _some_ Ruby along with the data, then we may not actually gain anything by splitting off the rest.

AFOliveira, 2025-03-03:
What about downstream usage? I see the point of not wanting all the tooling "messing" with your own tooling. Moreover, if someone already has there tools to use with riscv-opcodes  it would be much easier to sell the DB separately. If not before, I believe the summit will be great to poke people with the idea of trying to replace riscv-opcodes with UDB. If not for more, the lack of versioning must be something that haunts every downstream user of that repo.

ThinkOpenly, 2025-03-03:
I had a thought about this over the weekend, as one does.

Would it be useful to consider how UDB might be used when adding a new architecture, and splitting into a core "generic tools/infrastructure" component, and an arch-specific component? (This is only brainstorming at this point.)