riscv/riscv-unified-db #458: UDB json index

AFOliveira, 2025-02-11:
Last meeting, @drom requested a checked out index of the UDB in the repo, I have a similar question/idea. Can we turn the index configuration specfic? If we made an index of specific configuration in json, tools could also use it modularly, including @drom tool. Obviously we don't need to check out all indexes.

dhower-qc, 2025-02-11:
I'll be pushing a PR soon that adds an index during the resolution step. That means every configuration that you resolve will get an index (located in `gen/resolved_arch/<cfg name>/index.yaml`). I'm adding the resolved "unconfig" ("_") to the GitHub deployment, so there will always be an index reflecting `main` at https://riscv-software-src.github.io/riscv-unified-db/resolved_arch/index.yaml.

james-ball-qualcomm, 2025-02-11:
Can we choose a different name than "_" to mean unconfigured? That isn't obvious to new people what it means.

dhower-qc, 2025-02-11:
suggestion?

james-ball-qualcomm, 2025-02-11:
How about "complete_isa"?

james-ball-qualcomm, 2025-02-11:
Or "complete_arch"

drom, 2025-02-11:
I need just `index.json` "non resolved" version of index.

dhower-qc, 2025-02-12:
how about "rv"? We already have "rv32" and "rv64"