riscv/riscv-unified-db #464: The custom CSR prefix separated from CSR name by dot does not work

ayosher, 2025-02-19:
**Describe the bug**

The custom CSR prefix separated from CSR name by dot does not work:

**To Reproduce**
1. On branch https://github.com/riscv-software-src/riscv-unified-db/tree/Xqciv0p7
2. Run: ./do clean && ./do gen:ext_pdf EXT=Xqci VERSION=latest CFG=qc_iu

**Text output/Error Messages**
-------------------------------------------------------------------------------
/local/mnt/workspace/quic/riscv-unified-db/lib/idl/ast.rb:297:in `type_error': In file gen/arch/qc_iu/inst/Xqci/qc.c.clrint.yaml (Idl::AstNode::TypeError)
On line 28
In the code:

  27: Bits<12> MCLICIP0_ADDR = CSR[qc.mclicip0].address();
  28: 
  29: XReg idx = rs1 / 32;
  30: X

A type error occurred
  no symbol named 'qc' on line 28
-------------------------------------------------------------------------------

ayosher, 2025-02-19:
James, please have a look.

james-ball-qualcomm, 2025-02-19:
> WARNING: This email originated from outside of Qualcomm. Please be wary of any links or attachments, and do not enable macros.

I took a quick look at the issue. It looks like the problem is that the period isn't being converted to an underscore. This could probably be a pretty simple fix and I'll look into it.

+james+

james-ball-qualcomm, 2025-02-20:
The problem is that you've put "qc." in your IDL code (i.e., operation) instead of "qc_". You can't have a "." in a CSR name (just instruction names). I've modified the IDL code to fix this by searching for "CSR[qc." and replacing it with "CSR[qc_]". I'm rebuilding now to see how it works and if I've found all the problems yet.

james-ball-qualcomm, 2025-02-20:
Okay, that didn't fix it. Can you submit a GitHub issue against Derek saying UDB should create an error if it detects a CSR name with a "." in it and see what he says?

ayosher, 2025-02-20:
> Okay, that didn't fix it. Can you submit a GitHub issue against Derek saying UDB should create an error if it detects a CSR name with a "." in it and see what he says?

Probably I was not clear enough. CSR name **must** support dot in it (this explicit requirement we got from LLVM community). There is a requirement for custom CSR to have custom dot-separated prefix in its name. Currently UDB is not compliant with this requirement, practically making it impossible to create using UDB custom extensions that includes CSR (to be supported by LLVM). This should be fixed, IMHO.

james-ball-qualcomm, 2025-02-20:
I was afraid that was what you were saying yesterday. Well, UDB doesn't support that LLVM requirement at the moment. It will be up to Derek to access how/if this can be supported. The problem is that there aren't any RISC-V CSRs with a "." in their name because the "." is used to separate the CSR name from a CSR field name. So, having a "." in the CSR will break more tools than just UDB. 

Can we add Ana to this issue?

ayosher, 2025-02-20:
> Can we add Ana to this issue?
Sure, good idea. Logic behind the request is simple: any custom-extension instructions currently prefixed with company prefix, separated by "." - so the same request is for CSR.

apazos, 2025-02-20:
The CR naming rules are defined in the Toolchain Conventions spec, and the toolchains like GCC and LLVM implement them:
https://github.com/apazos/riscv-toolchain-conventions/blob/main/src/toolchain-conventions.adoc#csr-naming-scheme

In LLVM , internally we replace . with _ when auto generating C/C++ code to avoid errors with auto-generated macros. This is internal to the implementation, users are not impacted.

I suspect something similar will have to be done in the implementation of UDB. Will discuss it with Luke if Derek is not around.

ayosher, 2025-02-20:
Folks, think I succeed to overcome the issue and now things working for me - I will close this issue. Thank you for advices.
It turns out to bemisuse on my side - dot in names is possible, as far as I see now.

closed 2025-02-20 by ayosher