riscv/riscv-unified-db #470: Remove "Cached" information

AFOliveira, 2025-03-04:
Is there a easy way to set up an option that cleans all the outputs/cached items without needing to build or fetch singularity again?
Basically, something less powerful than `do clean` that would also be faster. 

This is clearly not a must, just a pet peeve I sometimes encounter and don't know how to solve :)

dhower-qc, 2025-03-04:
Should already do that:

do:
```bash
#!/bin/bash

ROOT=$(dirname $(realpath $BASH_SOURCE[0]))

# *** CLEAN HAPPENS WITHOUT CONTAINER ***
if [ "$1" == "clean" ]; then
  ${ROOT}/bin/clean
  exit $?
fi

source $ROOT/bin/setup

# really long way of invoking rake, but renamed to 'do'
$BUNDLE exec --gemfile $ROOT/Gemfile ruby -r rake -e "Rake.application.init('do');Rake.application.load_rakefile;Rake.application.top_level" -- $@

```

Are you seeing the container run even with `do clean`?

AFOliveira, 2025-03-05:
> Oh, I think I see what you are saying -- clean removes the container so the next command rebuilds?

Exactly! That alone ends up by taking up quite some adittional time when trying to play with outputs.

PR #491 (2025-03-21): Split Clean and Clobber into two different functions.
Closes #470 .
closed 2025-03-21 by AFOliveira