riscv/riscv-unified-db #473: lowest/highest_bit_set functions should return a Bits type?

ThinkOpenly, 2025-03-05:
**Describe the bug**
I think the IDL functions `lowest_set_bit` and `highest_set_bit` are defined slightly incorrectly.

In reviewing #471, the following assignment was being slightly corrected:
```
X[rd] = (xlen() - 1) - $signed(lowest_set_bit(X[rs1]));
```
- `X[rd]` and `X[rs1]` are of type `Bits[XLEN]` (I believe, as BUILTIN).
- `xlen()` is defined:
  ```
  function xlen {
    returns Bits<8>
  ```
  (I'll note that #471 is removing the use of `xlen()`, but it's useful for additional context here.)
- I confess not knowing where `$signed` is defined (?)
- `lowest_set_bit(...)` is defined:
  ```
  function highest_set_bit {
    returns XReg
    arguments XReg value
  [...]
  function lowest_set_bit {
    returns XReg
    arguments XReg value
  ```

It makes sense to me that the `*set_bit` functions accept an `XReg` parameter.

It does _not_ make sense to me that they return an `XReg` value. Ignoring `$signed`, the calculation is using register _values_ to compute an integer _value_ (less than or equal to XLEN).

Should these functions instead return Bits<8> (or essentially Bits<log2(XLEN)>, but we presume that 8 is sufficient)?

dhower-qc, 2025-03-05:
> I confess not knowing where $signed is defined (?)

It (and everything beginning with `$`) is builtin. It converts the expression to signed, which will result in a sign extension when there is a bits-widening conversion (e.g., when comparing the signed value to a wider bit vector).

> Should these functions instead return Bits<8> (or essentially Bits<log2(XLEN)>, but we presume that 8 is sufficient)?

They certainly could/should. Because the return value is unsigned, it doesn't make a functional difference unless something strange happens like the return value is used in a concatenation -- the return width is just bigger than it needs to be. FWIW, what you propose (Bits<log2(XLEN) _+1_>) will work as long as log2 argument is const (which XLEN is).

 **Regarding the statement that kicked this off**

| `X[rd] = (xlen() - 1) - $signed(lowest_set_bit(X[rs1]));`

`$signed` isn't needed.

ThinkOpenly, 2025-03-05:
> > I confess not knowing where $signed is defined (?)
> 
> It (and everything beginning with `$`) is builtin. It converts the expression to signed, which will result in a sign extension when there is a bits-widening conversion (e.g., when comparing the signed value to a wider bit vector).

Thanks. I found SystemVerilog reference, which I'll keep handy. ;-)

> They certainly could/should. Because the return value is unsigned, it doesn't make a functional difference unless something strange happens like the return value is used in a concatenation -- the [...]

I see, since `XReg` is `Bits<XLEN>`. It's a little confusing to be returning `XReg` since the return value is not associated with a register.

Leaving this open, as it could be a "good first issue".

dhower-qc, 2025-03-05:
Sounds good. FYI `XReg` is just an alias for `Bits<XLEN>`.

omsuneri, 2025-07-15:
@ThinkOpenly can you please assign this issue to me !!

commit 1b1692b4 (2025-09-17): FIXES ISSUE #473 : lowest/highest_set_bit functions should return a Bits type? (#906)

This PR fixes #473.
The ```lowest_set_bit``` and ```highest_set_bit``` functions were
returning XReg (i.e., Bits<XLEN>) though they only compute a position
index. Updated to return Bits<8> instead, which is sufficient and
semantically correct.
All smoke tests and IDL validations pass successfully.
@ThinkOpenly please review this and request if any further chnaged are
required !!
PR #906 (2025-09-17): FIXES ISSUE #473 : lowest/highest_set_bit functions should return a Bits type?
This PR fixes #473.
The ```lowest_set_bit``` and ```highest_set_bit``` functions were returning XReg (i.e., Bits<XLEN>) though they only compute a position index. Updated to return Bits<8> instead, which is sufficient and semantically correct.
All smoke tests and IDL validations pass successfully.
@ThinkOpenly please review this and request if any further chnaged are required !!