riscv/riscv-unified-db #478: Is RORW wrong?

AFOliveira, 2025-03-07:
RORW has the following variables in UDB:

```
  variables:
    - name: shamt
      location: 24-20
    - name: rs1
      location: 19-15
    - name: rd
      location: 11-7
```

However, the ISA is states that it's rs2 and not shamt. 

[image]

Aparently, sail code has shamt variable, but fetches it from rs2:
```
sail(): |
  {
    let rs1_val = (X(rs1))[31..0];
    let shamt = (X(rs2))[4..0];
    let result : bits(32) = match op {
      RISCV_ROLW => rs1_val <<< shamt,
      RISCV_RORW => rs1_val >>> shamt
    };
    X(rd) = sign_extend(result);
    RETIRE_SUCCESS
  }

```

I did not PR the change because it also requires IDL code and likely there are a lot of people who may change that with way less effort than me.

AFOliveira, 2025-03-13:
In PR #300 I am trying to match riscv-opcodes correctly, but there are some decisions to take and questions to made, like the one above. Another one is:

1. On csrrci and csrrsi, there is a variable name _uimm_ ; however, unsigned is something that should be explained on a subsfield, right?

AFOliveira, 2025-03-13:
@ThinkOpenly @dhower-qc I think you guys may have the answer to this questions

ThinkOpenly, 2025-03-17:
There are indeed some problems in RORW.

First, `shamt` is for _immediate_ values, not registers.

RORW uses a register, as can be seen in the Sail code. So, the variable name should be "rs2".

The IDL has related issues:
```
  XReg rs1_word = X[rs1][31:0];
  XReg shamt = X[rs1][4:0];
```

The second line is wrong. `shamt` should come from the lower 5 bits of _`rs2`_, not `rs1`.

ThinkOpenly, 2025-03-17:
@AFOliveira , if you're not in a hurry here, let's open an issue and let one of the new mentees address it, OK?

dhower-qc, 2025-03-17:
rorw:

In this case, I think we should change the YAML to use a register specifier instead of shamt. We use shamt when the encoding holds an immediate, but that isn't the case here. Note that changing this will also require changing the variable in IDL (from shamt -> rs2).

> On csrrci and csrrsi, there is a variable name uimm ; however, unsigned is something that should be explained on a subsfield, right?

Correct-ish. Right now, every variable is unsigned. If it needs to be signed, then it must be casted (using `$signed()`) in IDL. That said, the name of the field is just a name. `uimm` is more descriptive than `imm` so I'm not opposed to it. @ThinkOpenly, any thoughts?

AFOliveira, 2025-03-17:
> @AFOliveira , if you're not in a hurry here, let's open an issue and let one of the new mentees address it, OK?

Sure, I also have some other ideas for mentees. I'll send them to you and Derek tomorrow, so you can then choose them or not

ThinkOpenly, 2025-03-17:
@Shehrozkashif, mentioning you here, so I (hope I) can add you as assignee.

ThinkOpenly, 2025-03-17:
I still can't. I think you need to comment on this issue to be eligible to be an assignee.

AFOliveira, 2025-03-18:
> Correct-ish. Right now, every variable is unsigned. If it needs to be signed, then it must be casted (using `$signed()`) in IDL. That said, the name of the field is just a name. `uimm` is more [...]

This highlights a bigger problem: there's a inconsistency throughout the UDB. This is something we should take action on.
 
> We probably need a schema change that adds a "type" for the "variables", but that's a much bigger change. We're kind of kludging it now with special variable names, but explicit types would be better.

While kludging works for now, it creates problems when generating outputs from UDB. I'll open a more detailed issue tracking this.

### Proposed Solution
This could be a good task for a mentee:
1. First phase: Analyze variable type information requirements and propose a schema
2. Second phase: Structure the information for all variables
3. Third Phase: Structure into variable groups, likely with some referencing, as @dhower-qc has been proposing for a while now

While the Second and Third Phase may require some structural changes to the UDB and should be closely monitored, the First one is of high impact and will make the mentee study/learn a lot about RISC-V in general and about variables. Likely the First phase would then be presented in UDB's weekly meeting for feedback when it is close to an end. This appears to be a long-standing issue that no one has had bandwidth to solve, but it would have high impact.

@ThinkOpenly @dhower-qc please give me your opinion on it, I may write up a better structured scope of work and instructions so someone may pick it up.

Unique-Usman, 2025-03-18:
Hello, this seems like an interesting issues, but, as I am completely new, more pointers will be very helpful. I will go through it also and drop any question if I got any.

AFOliveira, 2025-03-18:
> Hello, this seems like an interesting issues, but, as I am completely new, more pointers will be very helpful. I will go through it also and drop any question if I got any.

Hi @Unique-Usman , thank you for your interest! I'll just get discuss this with Derek and Paul, and then post it in a much more explicit way, we may even set up a call if you prefer!

Unique-Usman, 2025-03-18:
Thanks @AFOliveira, I am definitely okay with a call also.

Shehrozkashif, 2025-03-18:
@ThinkOpenly Hi, I've updated the commit message as you suggested. Let me know if you need any further changes in the PR.

dhower-qc, 2025-03-18:
@Unique-Usman, glad to see this looks interesting. This is a big impact problem to solve. 

Paul gave a good overview of the current state of things. @AFOliveira has a good high-level outline of the task. Digging in a bit more for phase 1:

- We want to transition from storing the encoding information directly in the instruction object (under the `encoding` key) to storing _pointer_ to an encoding object that can be shared by multiple instructions.
    - Specifically, we want to use [JSON Reference](https://www.ietf.org/archive/id/draft-pbryan-zyp-json-ref-03.txt) to point somewhere else.
- A critical first step is to identify instructions that share the same set of information and belong in the same "bucket".
    - For example, the `add` and `sub` instructions have the same format, with the same variable names, just a different opcode.
    - Note that some instructions look the same but differ enough that we want to put them in separate buckets. For example, load instructions (like `lw`) have the same I-type format as immediate arithmetic instructions (like `addi`), but will call their immediate different things (offset vs. immediate) and will have different linker relocations.

A good first deliverable could be a script that sorts instructions into buckets. If the script includes an "unclassified" bucket, we can use it to track what's left, and iterate until we are happy. Here's a set of buckets off the top of my head:

- Basic R-type (e.g., add, sub). Has one X destination register and two X source registers. No relocation.
- Basic I-type (e.g., addi, ori) Has one X destination register, one X source register, a 12-bit immediate, and no relocation.
- Load type (e.g., lh, lw) Like I-type, but the immediate is called 'offset' and has a PC-relative relocation.
- JALR type: Like I-type, but immediate is called offset.
- Basic B-type (e.g., BLT, BEQ) Has two X source registers, a 12-bit shifted offset, and a PC-relative relocation.
- Basic S-type (e.g., sh, sw) Has two X source registers, a 12-bit offset, and a PC-relative relocation
- AUPIC (e.g., lui) Has one X source register, a 20-bit offset, and PC-relative relocation
- LUI (e.g., lui) Has one X source register, and a 20-bit immediate
- Basic J-type (e.g., jal) Has one X destination register, a 19-bit shifted offset, and a PC-relative relocation
- ... All the various C instruction types ...

AFOliveira, 2025-03-18:
To add to what @dhower-qc just mentioned, I would also say, check PR #453 in which a good amount of that work for a deliverable is done, maybe fork it and start from there if you want to create a script to identify instructions by types.

Shehrozkashif, 2025-03-18:
I agree that updating the schema to explicitly define variable types would help prevent inconsistencies in the long run. This change would make it clearer whether a value should be treated as `imm`, `uimm`, or another type, reducing ambiguity across different implementations. I’m happy to contribute to this discussion and help refine the schema accordingly.

Unique-Usman, 2025-03-18:
Thanks @ThinkOpenly and @dhower-qc, I will go through all this.

AFOliveira, 2025-03-19:
Back to this @ThinkOpenly @dhower-qc :

> Correct-ish. Right now, every variable is unsigned. If it needs to be signed, then it must be casted (using `$signed()`) in IDL. That said, the name of the field is just a name. `uimm` is more [...]

Given what is stated here, shouldn't it be switched to imm? If everything is unsigned, it just makes up for some confusion. Unless, we agree in calling uimm to all unsigned immediates and switch all the others. I don't oppose to any of the approaches, but removing the "u" would be much quicker - likely this is a problem that will need to be addressed in the near future anyways.

dhower-qc, 2025-03-19:
My preference would be to move everything to 'uimm' where it applies, since that is more descriptive. Also more work ;)

AFOliveira, 2025-03-19:
> > [@dhower-qc](https://github.com/dhower-qc) opinion on C extension variable "rd/rs1"? Can we just call it "rd/rs1" on UDB? It is significantly different from a default "rd".
> 
> What's the context for this question?

@ThinkOpenly I will move that discussion here, given that I messed up and only wrote there by accident. To identify variables from riscv-opcodes I always need to play around and try to get some identification. In the C extension, risc-v opcodes calls that register rd_rs1 and is the only difference from the standard rd. Also, the spec itself also calls this variable rd/rs1 given it actually is both.

Unique-Usman, 2025-03-19:
I have gone through the issues and explanations and I have understand fairly the requirements and what is intended to be done. 

> * AUPIC (e.g., lui) Has one X source register, a 20-bit offset, and PC-relative relocation
As @dhower-qc mentioned, I will be starting here. And also, as @AFOliveira mentioned, I will be going through the PR #453 which has done some quite amount of large work on it on this specific task.

AFOliveira, 2025-03-19:
> I have gone through the issues and explanations and I have understand fairly the requirements and what is intended to be done.
>
>
> As [@dhower-qc](https://github.com/dhower-qc) mentioned, I will be [...]

Nice! Please ask for help if/when you need it!

PR #521 (2025-03-20): Fix RORW operand and IDL
The YAML for the RORW instruction has shamt as an operand. shamt is used as an immediate value, but the RORW instruction uses a register value. The RISC-V ISA specification shows the analogous RORW instruction's operand is named rs2, which is appropriate here.

Change RORW operand shamt to rs2.

Also, in the IDL for RORW, the shamt _value_ should be retrieved from rs2 (not rs1, as is done currently.

Fix the retrieval of the shamt value to come from rs2. 
closes #478
closed 2025-03-20 by dhower-qc