riscv/riscv-unified-db #493: `bin/setup` doesn't work on MacOS  error "realpath: bin/setup[0]: No such file or directory"

kbroch-rivosinc, 2025-03-14:
**Describe the bug**
Unhappy from beginning trying to establish ROOT variable:

```
kbroch@kbroch-mac.local:~/rvi/repos/riscv-software-src/riscv-unified-db on  main via  v22.13.1 via 🐍 v3.13.1 via 💎 v3.2.3
❯ bin/setup
realpath: bin/setup[0]: No such file or directory
usage: dirname string [...]
usage: dirname string [...]
cat: /bin/.container-tag: No such file or directory
bin/setup: line 37: [: -v: unary operator expected
bin/setup: line 41: [: -v: unary operator expected
bin/setup: line 43: [: -v: unary operator expected
UDB tools run in a container. Both Docker and Singularity/Apptainer are supported.

1. Docker
2. Singularity

Which would you like to use? (1/2)
1
bin/setup: line 63: /.container-type: Read-only file system
Using docker environment
bin/setup: line 68: [: -v: unary operator expected
Building Docker image...
[+] Building 0.0s (0/0)                                                                                                                docker:desktop-linux
ERROR: invalid tag "riscvintl/udb:": invalid reference format
mkdir: /.home: Read-only file system
docker: Error response from daemon: the working directory 'riscvintl/udb:' is invalid, it needs to be an absolute path.
See 'docker run --help'.
docker: Error response from daemon: the working directory 'riscvintl/udb:' is invalid, it needs to be an absolute path.
See 'docker run --help'.
docker: Error response from daemon: the working directory 'riscvintl/udb:' is invalid, it needs to be an absolute path.
See 'docker run --help'.
docker: Error response from daemon: the working directory 'riscvintl/udb:' is invalid, it needs to be an absolute path.
See 'docker run --help'.
docker: invalid reference format.
See 'docker run --help'.
bin/setup: line 151: /.home/.venv/bin/activate: No such file or directory
docker: invalid reference format.
See 'docker run --help'.
docker: Error response from daemon: the working directory 'riscvintl/udb:' is invalid, it needs to be an absolute path.
See 'docker run --help'.
bin/setup: line 209: /.git/hooks/pre-commit: No such file or directory
chmod: /.git/hooks/pre-commit: No such file or directory
```

**To Reproduce**

```
kbroch@kbroch-mac.local:~/rvi/repos/riscv-software-src/riscv-unified-db on  main via  v22.13.1 via 🐍 v3.13.1 via 💎 v3.2.3
❯ /bin/bash --version
GNU bash, version 3.2.57(1)-release (arm64-apple-darwin24)
Copyright (C) 2007 Free Software Foundation, Inc.

kbroch@kbroch-mac.local:~/rvi/repos/riscv-software-src/riscv-unified-db on  main via  v22.13.1 via 🐍 v3.13.1 via 💎 v3.2.3
❯ sw_vers
ProductName:            macOS
ProductVersion:         15.3.1
BuildVersion:           24D70

kbroch@kbroch-mac.local:~/rvi/repos/riscv-software-src/riscv-unified-db on  main via  v22.13.1 via 🐍 v3.13.1 via 💎 v3.2.3
❯ $SHELL --version
zsh 5.9 (arm64-apple-darwin24.0)
```

kbroch-rivosinc, 2025-03-14:
ran shellcheck and found one error (lots of warning so turned that off for now):
NOTE: that error isn't related to this problem.

```
kbroch@kbroch-mac.local:~/rvi/repos/riscv-software-src/riscv-unified-db on  main via  v22.13.1 via 🐍 v3.13.1 via 💎 v3.2.3 took 8s
❯ shellcheck -S error bin/setup

In bin/setup line 162:
    if [ ! -d "${ROOT}/.home/.yard/gem_index"]; then
       ^-- SC1073 (error): Couldn't parse this test expression. Fix to allow more checks.
              ^-- SC1019 (error): Expected this to be an argument to the unary condition.
                                              ^-- SC1020 (error): You need a space before the ].
                                              ^-- SC1072 (error): Missing space before ]. Fix any mentioned problems and try again.

For more information:
  https://www.shellcheck.net/wiki/SC1020 -- You need a space before the ].
  https://www.shellcheck.net/wiki/SC1019 -- Expected this to be an argument t...
  https://www.shellcheck.net/wiki/SC1072 -- Missing space before ]. Fix any m...
```

kbroch-rivosinc, 2025-03-14:
after fixing the following: 

```diff
diff --git a/bin/setup b/bin/setup
index 57776bf3..efd780f2 100755
--- a/bin/setup
+++ b/bin/setup
@@ -1,7 +1,8 @@
 #!/bin/bash

-ROOT=$(dirname $(dirname $(realpath $BASH_SOURCE[0])))
+ROOT=$(dirname $(dirname $(realpath ${BASH_SOURCE[0]})))
 OLDWD=$PWD
+echo -e "cding to $ROOT"
 cd $ROOT

 CONTAINER_TAG=`cat ${ROOT}/bin/.container-tag`
@@ -159,7 +160,7 @@ fi
 # fi

 if [[ ! -z "$DEVELOPMENT" && $DEVELOPMENT -eq 1 ]]; then
-    if [ ! -d "${ROOT}/.home/.yard/gem_index"]; then
+    if [ ! -d "${ROOT}/.home/.yard/gem_index" ]; then
         ${RUN} bundle exec --gemfile ${ROOT}/Gemfile yard config --gem-install-yri
         ${RUN} bundle exec --gemfile ${ROOT}/Gemfile yard gems
         touch ${ROOT}/.stamps/dev_gems
```

I see this now: 

```
kbroch@kbroch-mac.local:~/rvi/repos/riscv-software-src/riscv-unified-db on  main [!] via  v22.13.1 via 🐍 v3.13.1 via 💎 v3.2.3 took 44s
❯ ./bin/setup
cding to /Users/kbroch/rvi/repos/riscv-software-src/riscv-unified-db
./bin/setup: line 38: [: -v: unary operator expected
Using docker environment
./bin/setup: line 69: [: -v: unary operator expected

```

version of `/bin/bash` with macos:

```
kbroch@kbroch-mac.local:~/rvi/repos/riscv-software-src/riscv-unified-db on  main [!] via  v22.13.1 via 🐍 v3.13.1 via 💎 v3.2.3
❯ /bin/bash --version
GNU bash, version 3.2.57(1)-release (arm64-apple-darwin24)
Copyright (C) 2007 Free Software Foundation, Inc.
```

Seems ancient (5.x is typical) and what I have in path from homebrew:

```
kbroch@kbroch-mac.local:~/rvi/repos/riscv-software-src/riscv-unified-db on  dev/kbroch/fix-setup-macos:main [!⇡] via  v22.13.1 via 🐍 v3.13.1 via 💎 v3.2.3 took 29s
❯ /usr/bin/env bash --version
GNU bash, version 5.2.37(1)-release (aarch64-apple-darwin23.4.0)
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

This is free software; you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

So allow script to pick bash from env: 

```diff
diff --git a/bin/setup b/bin/setup
index efd780f2..bae0d51f 100755
--- a/bin/setup
+++ b/bin/setup
@@ -1,4 +1,4 @@
-#!/bin/bash
+#!/usr/bin/env bash

 ROOT=$(dirname $(dirname $(realpath ${BASH_SOURCE[0]})))
 OLDWD=$PWD
```

Seems to be working now: 

```
kbroch@kbroch-mac.local:~/rvi/repos/riscv-software-src/riscv-unified-db on  dev/kbroch/fix-setup-macos:main [!⇡] via  v22.13.1 via 🐍 v3.13.1 via 💎 v3.2.3 took 11s
❯ ./bin/setup
cding to /Users/kbroch/rvi/repos/riscv-software-src/riscv-unified-db
Using docker environment
```

kbroch-rivosinc, 2025-03-14:
@dhower-qc : unless I'm still confused, I've come to the realization that `bin/setup` (and all the scripts) only runs from inside a container already?

If that is the case, I can see how some of these issue because less important, but I would still recommend they be change to make thing more resilient to container base changes or support native development.

dhower-qc, 2025-03-17:
Correct; the files under `./bin` run in the selected container environment**.

@kbroch-rivosinc, yes, we should still clean these files up. 

@AFOliveira, we could add a native option for the environment (that would just use an empty container prefix on the commands), but I suggest we make it clear that it's a 100% unsupported option.

** Except setup, which sets up the container environment, and clean, which just removes generated files without depending on the container.

kbroch-rivosinc, 2025-03-18:
Thanks for responses.  I took out the shellcheck errors and put fixes in #500.  

I still think there's use in updating the script to use `#!/usr/bin/env bash` from a good practices standpoint, but don't feel strongly about this.

dhower-qc, 2025-03-19:
No drawback, potential improvement. 👍

AFOliveira, 2025-03-19:
Is this solved by #500 or only partially solved?

> @AFOliveira, we could add a native option for the environment (that would just use an empty container prefix on the commands), but I suggest we make it clear that it's a 100% unsupported option.

Should we update the README?

> I still think there's use in updating the script to use `#!/usr/bin/env bash` from a good practices standpoint, but don't feel strongly about this.

Was this only for `do `or do you think it should be done in all?

AFOliveira, 2025-03-19:
@dhower-qc opinion on C extension variable "rd/rs1"? Can we just call it "rd/rs1" on UDB? It is significantly different from a default "rd".

dhower-qc, 2025-03-19:
We'll have to call it something else; rd/rs1 would be an illegal IDL name. rd_rs1 would work.

ThinkOpenly, 2025-03-19:
> [@dhower-qc](https://github.com/dhower-qc) opinion on C extension variable "rd/rs1"? Can we just call it "rd/rs1" on UDB? It is significantly different from a default "rd".

What's the context for this question?

kbroch-rivosinc, 2025-03-21:
Ok #494 landed so I think everything is done here.

commit 297bfa81 (2025-03-20): use env to select bash (#494)

fixes #493
closed 2025-03-21 by kbroch-rivosinc