riscv/riscv-unified-db #497: Add Zilsd extension

AFOliveira, 2025-03-17:
**Describe the data**
Zilsd extensions has been recently ratified and at some point needs to be added to UDB.
https://github.com/riscv/riscv-zilsd/releases/tag/v1.0

AFOliveira, 2025-03-17:
@christian-herber-nxp  Would you be willing to commit your extension to UDB?

christian-herber-nxp, 2025-03-17:
I am not yet familiar with the unified DB and how add an extension. It will be quicker if somebody familiar does it. 
Especially as I assume that this requires more than just adding the yank, but to introduce the new load/store behavior

AFOliveira, 2025-03-17:
I'll keep the issue open to keep progress of adding the extension. Anyways, thanks for the quick answer.

dhower-qc, 2025-03-17:
@latifbhatti this could be a good starter issue.

latifbhatti, 2025-03-18:
Sounds great! I'll take a look and get started on it. Thanks for assigning the task!

dhower-qc, 2025-03-19:
Split into #510, #511

closed 2025-03-19 by dhower-qc