riscv/riscv-unified-db #499: Add Contributing guidelines

dhower-qc, 2025-03-17:
**Is your feature request related to a problem? Please describe.**
We are missing a Contributing document.

**Describe the solution you'd like**
We need to add a document describing our process to contribute code. What exactly is that process? TBD, to be discussed in this issue.

dhower-qc, 2025-03-17:
@kersten1, I was hoping you could contribute to this issue as well.

dhower-qc, 2025-03-17:
Off the top of my head, we should add:

- How to report a bug / missing data.
- PR review policy
    - CODEOWNERS
- Issues / assignment
- Commit message expectations / formatting
- Coding conventions (pointing them to linting tool configs)
- Regression requirements

kersten1, 2025-03-18:
- Provide a template to follow - for both the overall spec and the yaml itself.
- Sample yaml code
- yaml linter? Explain how to add one to vscode as well as run one against PRs.

ThinkOpenly, 2025-04-01:
For commits, there's a lot of subjectivity. GCC is on the stricter side: https://gcc.gnu.org/contribute.html.

Of particular interest:
```
A high-quality e-mail subject line for a contribution contains the following elements:

- A classifier
- Component tags
- An optional series identifier
- A brief summary
- An optional bug number

The entire line (excluding the classifier) should not exceed 75 characters.
```

But GCC is not a GitHub-based project.

For UDB, we could consider components, but not sure it's worth it at present.

Really, all we probably need is a brief summary, and maybe a length limit.

For the brief summary, it's nice (more readable, understandable) if they are fairly consistent. I'll throw some guidelines, based on my experiences:
- Imperative tense ("Fix [W]", "Refactor [X]", "Add [Y]", "Move code [Z]")
- No issue numbers, no explanation, no details, no caused by. Just "what does this commit do?"

For the commit message, if the commit subject is not self-explanatory (most aren't):
- Describe the issue, including example with incorrect results and actual output if possible.
- Describe how the issue was fixed, including correct results/output if possible.
- Any additional testing done, although we ask for regression to be run, so maybe that doesn't need to be described here.

For bug fixes, should we encourage/mandate a new regression test?

The hard part, especially for GitHub projects: do we want to be particular about commits being fairly pristine and non-breaking? Currently, if there's, say, a typo in the first commit of many in a PR, then it's easiest for everyone to just add yet another commit that fixes the typo. However, the commit history preserves the original typo forever (absent a squash commit).

Is that OK, or should we try to keep things pristine? This is much more important for complex projects (absent CI as well) where it can be difficult to find bugs during overall testing and bisecting the commit history is incredibly helpful, and a bad commit can screw that up even if a slightly later commit fixed it. Maybe we don't need that level of hygiene here.

ThinkOpenly, 2025-04-01:
Just thinking... we could define the preferred process for submitting pull requests:
- fork the repo
- create a topic branch in that fork
- do all work on that topic branch
- create a merge request from that into UDB main

dhower-qc, 2025-04-02:
@ThinkOpenly, are you familiar with https://www.conventionalcommits.org/en/v1.0.0/?

It seems like a reasonable, and useful, set of guidelines. It has a mechanism for components like you mention. There is a tool (commitlint) that will check formats, and that we can use in CI. Good overview here: https://www.satellytes.com/blog/post/writing-and-enforcing-conventional-commit-messages-and-pull-request-titles.

ThinkOpenly, 2025-04-03:
Seems reasonable, and not overbearing. I love that it can be automated. It doesn't replace review work, but would surely help.

closed 2025-04-28 by dhower-qc