riscv/riscv-unified-db #505: Add Zvqdotq extension

AFOliveira, 2025-03-19:
Add Zvqdotq extension

dhower-qc, 2025-03-19:
This is still in development. I propose we table it for now, unless @aswaterman or another involved in the spec is willing to keep UDB updated as it goes.

aswaterman, 2025-03-19:
@kdockser is pushing this extension through the process, so I'll let him make the call.

kdockser, 2025-03-19:
I'd be beyond thrilled to add it to the UDB.
Please let me know what I need to dol.

AFOliveira, 2025-03-20:
Hello @kdockser, nice to hear those words! Welcome! To add your extension to UDB, you will need to roughly follow this steps:

In general, what you need to do is all under the `arch/` directory:

- First, go to `arch/ext` folder and  you will need to do a scheam like the following:

```
# yaml-language-server: $schema=../../schemas/ext_schema.json

$schema: "ext_schema.json#"
kind: extension
name: Zvqdotq
type: either _unprivileged or privileged_ 
long_name: _This is the long "english" name of the extension_
company:
  name: _RISC-V International_ - You may switch this for Tenstorrent, but I suppose that if it goes for ratification, it will belong to RVI
  url: _https://riscv.org_
doc_license:
  name: Creative Commons Attribution 4.0 International License
  url: https://creativecommons.org/licenses/by/4.0/
versions:
  - version: _"0.0.0"_
    state: _unratified_
    ratification_date: _2024-04_
    contributors:
      - name: _Kenneth Dockser_
        email: 
        company: 
    url: _https://drive.google.com/file/d/1SgLoasaBjs5WboQMaU3wpHkjUwV71UZn/view_
    implies:
      - [extension_A, "1.0.0"]
      - [extension_B, "1.0.0"]
description: |
This will be your textual description of the extension - just write plain asciidoc as you would normally here, it will then be converted.

params: //any hardware parameter you may want to define
  MUTABLE_MISA_B:
    description: |
      Indicates whether or not the `B` extension can be disabled with the `misa.B` bit.
    schema:
      type: boolean
```

Afterwards, you will need to write `arch/csr` and `arch/inst`- in these you may define all CSRs and Instructions that your extension defines each in a separate yaml, like `arch/inst/Zvqdotq/instructionC.yaml ` or `arch/csr/Zvqdotq/CSR_cycles.yaml`.

An example for instructions you can check per example the Add instruction, it should be fairly easy to adapt. There are two fields in the yaml that are Sail and IDL - and these are completely optional, so I would say just to leave them empty, someone may take care of that in the future.
```

$schema: "inst_schema.json#"
kind: instruction
name: add
long_name: Integer add
description: |
  Add the value in rs1 to rs2, and store the result in rd.
  Any overflow is thrown away.
definedBy: I
assembly: xd, xs1, xs2
encoding:
  match: 0000000----------000-----0110011
  variables:
    - name: rs2
      location: 24-20
    - name: rs1
      location: 19-15
    - name: rd
      location: 11-7
access:
  s: always
  u: always
  vs: always
  vu: always
data_independent_timing: true
operation(): |
leave empty

sail(): |
  leave empty
```

and for CSRs, it is slightly more complicated, but should look something like:

```
# yaml-language-server: $schema=../../schemas/csr_schema.json

$schema: "csr_schema.json#"
kind: csr
name: mcycle
long_name: Machine Cycle Counter
definedBy: Zicntr
address: 0xB00
description: |
  Counts the number of clock cycles executed by the processor core on which
  the hart is running.
  The counter has 64-bit precision on all RV32 and RV64 harts.

  The `mcycle` CSR may be shared between harts on the same core, in which case
  writes to `mcycle` will be visible to those harts. The platform should provide
  a mechanism to indicate which harts share an `mcycle` CSR.
priv_mode: M
length: 64
fields:
  COUNT:
    alias:
      - cycle.COUNT
    location: 63-0
    type: RW-RH
    sw_write(csr_value): |
      # since writes to this register may not be hart-local, it must be handled
      # as a special case
      if (xlen() == 32) {
        return sw_write_mcycle({read_mcycle()[63:31], csr_value.COUNT[31:0]});
      } else {
        return sw_write_mcycle(csr_value.COUNT);
      }
    description: |
      Cycle counter.

      <%- if ext?(:Zicntr) -%>
      Aliased as `cycle.CYCLE`.
      <%- end -%>

      Increments every cycle unless:

        * `mcountinhibit.CY` <%- if ext?(:Smcdeleg) -%>or its alias `scountinhibit.CY`<%- end -%> is set
        <%- if ext?(:Smcntrpmf) -%>
        * `mcyclecfg.MINH` is set and the current privilege level is M
        <%- if ext?(:S) -%>
        * `mcyclecfg.SINH` <%- if ext?(:Ssccfg) -%>or its alias `instretcfg.SINH`<%- end -%> is set and the current privilege level is (H)S
        <%- end -%>
        <%- if ext?(:U) -%>
        * `mcyclecfg.UINH` <%- if ext?(:Ssccfg) -%>or its alias `instretcfg.SINH`<%- end -%> is set and the current privilege level is U
        <%- end -%>
        <%- if ext?(:H) -%>
        * `mcyclecfg.VSINH` <%- if ext?(:Ssccfg) -%>or its alias `instretcfg.SINH`<%- end -%> is set and the current privilege level is VS
        * `mcyclecfg.VUINH` <%- if ext?(:Ssccfg) -%>or its alias `instretcfg.SINH`<%- end -%> is set and the current privilege level is VU
        <%- end -%>
        <%- end -%>
    reset_value: UNDEFINED_LEGAL
    affectedBy: [Zicntr, Smcntrpmf, Smcdeleg, Ssccfg]
sw_read(): |
  # you may leave this empty

```

In summary, I would say to start with `arch/ext` and `arch/inst`, since they should be fairly easy to write, specially, if you look at other extensions since it is just adapting what is already there, but please just ask more questions, if you find any problems or if you find my explanation not good  enough to start with. Likely @dhower-qc may provide a better one :)

dhower-qc, 2025-03-20:
@AFOliveira, let's sign you up for documentation ;)

ThinkOpenly, 2025-03-20:
> I was hoping this would be something we may later point people to, maybe even serve as the root for the guide on "How to get started on UDB."

Good idea! We don't even have a `doc` directory yet... great place for a developer's guide.

7908837174, 2025-07-16:
> @7908837174 <https://github.com/7908837174>.
> <riscv-software-src/riscv-unified-db/issue/505/issue_event/18640198114@
> github.com>

Thanks