riscv/riscv-unified-db #508: Add YAML files for all missing extensions

dhower-qc, 2025-03-19:
**Describe the data**

We need to identify all missing extensions and add _at least_ a basic YAML file with:

- Name
- Long name
- Version(s)

dhower-qc, 2025-03-19:
@AFOliveira, do you have any ideas for an existing list that we can compare against to find all the extensions that we haven't added (just as placeholders) in UDB?

AFOliveira, 2025-03-19:
> [@AFOliveira](https://github.com/AFOliveira), do you have any ideas for an existing list that we can compare against to find all the extensions that we haven't added (just as placeholders) in UDB?

If they have instructions, we can use riscv-opcodes - this is what I just did. If they don't add any instruction, then I don't have a direct method