riscv/riscv-unified-db #513: Add S{m,s}ctr extension IDL

dhower-qc, 2025-03-19:
The recently ratified (Nov24) extension S{m,s}ctr Control Transfer Records needs IDL:

registers (m, s, vs)
1x new system instruction (SCTRCLR)

Zain2050, 2025-09-26:
@ThinkOpenly please assign this issue to me.

Zain2050, 2025-09-29:
I want to work on its IDL. When will PR #554 get merged. Should I create new yaml files for smctr/ssctr (to define config params) & sctrclr (to describe it in IDL) and leave out rest of the fields as they will be pushed by that PR?

ThinkOpenly, 2025-10-02:
> When will PR [#554](https://github.com/riscv-software-src/riscv-unified-db/pull/554) get merged.

Looking at it now, I think it just needs a rebase.

> Should I create new yaml files for smctr/ssctr (to define config params) & sctrclr (to describe it in IDL) and leave out rest of the fields as they will be pushed by that PR?

If you wanted to build on top of this PR, base your branch on the branch for this PR.