riscv/riscv-unified-db #514: Add Zcmop extension data

dhower-qc, 2025-03-19:
Add Zcmop.yaml file to database.

Shehrozkashif, 2025-03-19:
@dhower-qc 

I was reading the ISA and found this extension in the unprivileged ISA :
https://drive.google.com/file/d/1uviu1nH-tScFfgrovvFCrj7Omv8tFtkp/view

it includes the following instructions

[image]

so I have to add separate yaml files for all instructions?

dhower-qc, 2025-03-19:
Generally speaking, yes. But now that I'm looking, Zcmop already exists in the database (arch/ext/Zcmop.yaml). @AFOliveira, is it possible you were using an old repo when you checked for missing extensions?

AFOliveira, 2025-03-19:
> Generally speaking, yes. But now that I'm looking, Zcmop already exists in the database (arch/ext/Zcmop.yaml). [@AFOliveira](https://github.com/AFOliveira), is it possible you were using an old repo when you checked for missing extensions?

Not really, Zcmop exists as an extension in the UDB, but the instructions do not. They are in the description because they were copy pasted from the ISA, but they are not under arch/inst. I did not check for extensions and assumed that it was simply all missing, instead of just this tiny part. In this case I would say that we just need to remove the instructions from arch/ext/Zcmop.yaml description and add the instrucitons under arch/inst. 

Btw @Shehrozkashif If you are adding them, it may be better to add them in the same style as "mop.r.n" instead of all of them - to follow with current style

Shehrozkashif, 2025-03-20:
Understood I'll add them 👍

PR #530 (2025-04-09): YAML files for the eight 16-bit MOP instructions defined in the Zcmo…
This PR implements the **C.MOP.N (Compressed May-Be-Operation) instruction** as part of the Zcmop extension. The instruction follows a **16-bit encoding format**, where bits `[15:12]` are fixed as `"0110"`, bits `[7:0]` are set to `"10000001"`, and bits `[11:8]` encode `(n >> 1)`, determining the specific operation. It maps to various **pseudoinstructions** (`c.mop.1`, `c.mop.3`, `c.mop.5`, etc.) based on the `mop_n_field` value. The instruction is available across all privilege levels (`s`, `u`, `vs`, and `vu`), ensuring br [... cut]
closed 2025-04-09 by ThinkOpenly