riscv/riscv-unified-db #520: SpellCheck pre-commit requires Docker

dhower-qc, 2025-03-20:
**Describe the bug**
SpellCheck pre-commit requires Docker

**To Reproduce**
Steps to reproduce the behavior:
1. commit

**Text output/Error Messages**
ShellCheck v0.10.0.......................................................Failed
- hook id: shellcheck
- exit code: 1

Executable `docker` not found

@kbroch-rivosinc, do you know if it's possible to run without Docker? So far, we've claimed that you need _either_ Singularity or Docker.

ayosher, 2025-03-20:
@kbroch-rivosinc , could you please look at commit e246e8ac2e30ed7c9f4607e2c92ed9dcb64249dc on cpp_hart branch? I added there more checks for scripts in /bin directory, looks after your commit to main some stuff left not fixed...

kbroch-rivosinc, 2025-03-20:
> [@kbroch-rivosinc](https://github.com/kbroch-rivosinc) , could you please look at commit [e246e8a](https://github.com/riscv-software-src/riscv-unified-db/commit/e246e8ac2e30ed7c9f4607e2c92ed9dcb64249dc) on cpp_hart branch? I added there more fixes for scripts in /bin directory, looks after your commit to main some stuff left not fixed... bin/clang-format bin/clang-tidy bin/clobber bin/g++ bin/gdb bin/make bin/rake bin/rdbg

I think the fix for shellcheck in is.  Once it lands in main you should be able to rebase and have a working singularity build.

commit 5720b70f (2025-03-20): change to local pre-commit hook for shellcheck

so can run in either container environment
fixes #520
commit 7098177c (2025-03-26): create local shellcheck hook that uses shellcheck from container

fixes #520
commit 0f11bddf (2025-03-26): create local shellcheck hook that uses shellcheck from container

fixes #520
PR #523 (2025-03-28): change to local pre-commit hook for shellcheck
so can run in either container environment
fixes #520

@dhower-qc : I'm not sure if I need anything else to get shellcheck installed in singularity container?
closed 2025-03-28 by dhower-qc