riscv/riscv-unified-db #527: Missing CSRs

AFOliveira, 2025-03-21:
While trying to generate RISC-V Go input I came across some missing CSRs. I am not super familiar with CSRs neither I took the time to really confirm that this is all missing, but from what I found, I think we are missing the following CSRs, is this right? I may be able to generate some kind of minimal skeleton out of riscv-opcodes.

```
  0x1 : "FFLAGS"
  0x2 : "FRM"
  0x7 : "UTVT"
  0x8 : "VSTART"
  0x9 : "VXSAT"
  0xa : "VXRM"
  0xf : "VCSR"
  0x11 : "SSP"
  0x15 : "SEED"
  0x17 : "JVT"
  0x45 : "UNXTI"
  0x46 : "UINTSTATUS"
  0x48 : "USCRATCHCSW"
  0x49 : "USCRATCHCSWL"
  0x102 : "SEDELEG"
  0x103 : "SIDELEG"
  0x104 : "SIE"
  0x107 : "STVT"
  0x10c : "SSTATEEN0"
  0x10d : "SSTATEEN1"
  0x10e : "SSTATEEN2"
  0x10f : "SSTATEEN3"
  0x120 : "SCOUNTINHIBIT"
  0x145 : "SNXTI"
  0x146 : "SINTSTATUS"
  0x148 : "SSCRATCHCSW"
  0x149 : "SSCRATCHCSWL"
  0x14d : "STIMECMP"
  0x14e : "SCTRCTL"
  0x14f : "SCTRSTATUS"
  0x150 : "SISELECT"
  0x151 : "SIREG"
  0x152 : "SIREG2"
  0x153 : "SIREG3"
  0x155 : "SIREG4"
  0x156 : "SIREG5"
  0x157 : "SIREG6"
  0x15c : "STOPEI"
  0x15f : "SCTRDEPTH"
  0x181 : "SRMCFG"
  0x204 : "VSIE"
  0x240 : "VSSCRATCH"
  0x244 : "VSIP"
  0x24d : "VSTIMECMP"
  0x24e : "VSCTRCTL"
  0x250 : "VSISELECT"
  0x251 : "VSIREG"
  0x252 : "VSIREG2"
  0x253 : "VSIREG3"
  0x255 : "VSIREG4"
  0x256 : "VSIREG5"
  0x257 : "VSIREG6"
  0x25c : "VSTOPEI"
  0x307 : "MTVT"
  0x308 : "MVIEN"
  0x309 : "MVIP"
  0x30c : "MSTATEEN0"
  0x30d : "MSTATEEN1"
  0x30e : "MSTATEEN2"
  0x30f : "MSTATEEN3"
  0x321 : "MCYCLECFG"
  0x322 : "MINSTRETCFG"
  0x345 : "MNXTI"
  0x346 : "MINTSTATUS"
  0x348 : "MSCRATCHCSW"
  0x349 : "MSCRATCHCSWL"
  0x34e : "MCTRCTL"
  0x350 : "MISELECT"
  0x351 : "MIREG"
  0x352 : "MIREG2"
  0x353 : "MIREG3"
  0x355 : "MIREG4"
  0x356 : "MIREG5"
  0x357 : "MIREG6"
  0x35c : "MTOPEI"
  0x5a8 : "SCONTEXT"
  0x603 : "HIDELEG"
  0x604 : "HIE"
  0x607 : "HGEIE"
  0x608 : "HVIEN"
  0x609 : "HVICTL"
  0x60c : "HSTATEEN0"
  0x60d : "HSTATEEN1"
  0x60e : "HSTATEEN2"
  0x60f : "HSTATEEN3"
  0x644 : "HIP"
  0x645 : "HVIP"
  0x646 : "HVIPRIO1"
  0x647 : "HVIPRIO2"
  0x6a8 : "HCONTEXT"
  0x7a0 : "TSELECT"
  0x7a1 : "TDATA1"
  0x7a2 : "TDATA2"
  0x7a3 : "TDATA3"
  0x7a4 : "TINFO"
  0x7a5 : "TCONTROL"
  0x7a8 : "MCONTEXT"
  0x7aa : "MSCONTEXT"
  0x7b0 : "DCSR"
  0x7b1 : "DPC"
  0x7b2 : "DSCRATCH0"
  0x7b3 : "DSCRATCH1"
  0xc20 : "VL"
  0xc21 : "VTYPE"
  0xc22 : "VLENB"
  0xda0 : "SCOUNTOVF"
  0xdb0 : "STOPI"
  0xe12 : "HGEIP"
  0xeb0 : "VSTOPI"
  0xfb0 : "MTOPI"
```

AFOliveira, 2025-03-21:
@james-ball-qualcomm @dhower-qc I think you guys are the ones who may be able to answer if I am correct more easily

AFOliveira, 2025-03-22:
Alright, I'll try to keep track of what we are missing here then, given the PRs that are coming in. 
For now, I'll tag #524

dhower-qc, 2025-03-27:
I have these tracked in issues, grouped by extension. They are tracked overall with the CSR phase 1 milestone.

I've left out the CLIC CSRs since they aren't ratified:

*  0x7 : "UTVT"
*  0x45 : "UNXTI"
*  0x46 : "UINTSTATUS"
*  0x48 : "USCRATCHCSW"
*  0x49 : "USCRATCHCSWL"
*  0x107 : "STVT"
*  0x145 : "SNXTI"
*  0x146 : "SINTSTATUS"
*  0x148 : "SSCRATCHCSW"
*  0x149 : "SSCRATCHCSWL"
*  0x307 : "MTVT"
*  0x345 : "MNXTI"
*  0x346 : "MINTSTATUS"
*  0x348 : "MSCRATCHCSW"
*  0x349 : "MSCRATCHCSWL"

closed 2025-03-27 by dhower-qc