riscv/riscv-unified-db #557: Add Smcsrind/Sscsrind YAML

dhower-qc, 2025-03-27:
Add extension files for Smcsrind/Sscsrind

syedowaisalishah, 2025-03-28:
I would like to work on this issue. Can you assign it to me?

syedowaisalishah, 2025-03-31:
@dhower-qc I’ve added the extension YAML files for Smcsrind and Sscsrind. For the CSR YAML files, could you recommend a structure that I should follow? Additionally, should I create separate folders for MISELECT, SISELECT, and VSISELECT CSRs?

dhower-qc, 2025-04-01:
> could you recommend a structure that I should follow?

Any YAML file under `arch/csr` will have the general structure. The key thing is to fill in the correct data for the CSR (like address) and then a list of fields (which must be present, even if there is a single 'field' for the entire CSR).

> should I create separate folders for ex

We should create a `arch/csr/Smcsrind` folder and add all the CSR yaml files there.

syedowaisalishah, 2025-04-01:
@dhower-qc Thanks for the clarification! I’ll add all the Smcsrind CSR YAMLs under arch/csr/Smcsrind and make sure each file includes the correct address and fields, even if it's just one.

commit 34157aae (2025-05-05): Add Smcsrind and Sscsrind CSR YAML Files (#590)

This PR adds:
- Extension YAML files for `Smcsrind` and `Sscsrind`
- CSR YAMLs for:
  - MISELECT, MIREG to MIREG6
  - SISELECT, SIREG to SIREG6
  - VSISELECT, VSIREG to VSIREG6
  
  Closes #557

---------
PR #590 (2025-05-05): Add Smcsrind and Sscsrind CSR YAML Files
This PR adds:
- Extension YAML files for `Smcsrind` and `Sscsrind`
- CSR YAMLs for:
  - MISELECT, MIREG to MIREG6
  - SISELECT, SIREG to SIREG6
  - VSISELECT, VSIREG to VSIREG6
  
  Closes #557
closed 2025-05-05 by ThinkOpenly