riscv/riscv-unified-db #580: Can't find defined_by? anymore in DatabaseObject

james-ball-qualcomm, 2025-03-31:
The fork I have of UDB relies heavily on defined_by? but it is gone. What is the replacement?

AFOliveira, 2025-03-31:
@james-ball-qualcomm what defined_by do you mean any ruby object or the definition in the yaml? Seems like the yaml files is still in place 

https://github.com/riscv-software-src/riscv-unified-db/blob/20eeeff520eb2adf1b3609d32e8b4c3441f164ae/arch/inst/I/add.yaml#L10C1-L10C10

dhower-qc, 2025-03-31:
It's been removed (rather than modified) to force any use to change how it works. `defined_by` used to return an array of extensions that define the object, but we can't rely on something that simple because the conditions can be more complex.

You now need to use `defined_by_condition`.

closed 2025-03-31 by dhower-qc