riscv/riscv-unified-db #601: Clarify and Support n = (field_n[3:1] << 1) | 1 Field Transformation in Instruction Schema

Shehrozkashif, 2025-04-09:
While working with compressed instruction formats (e.g., `c.mop`), we encountered a case where only the top 3 bits of a 4-bit field `n` are stored in bits 10–8. To reconstruct the full value of `n`, the following transformation is applied:

```
n = (field_n[3:1] << 1) | 1
```

This ensures the resulting value is always odd, with the least significant bit set to 1.

 Problem:
- The current schema (IDL) doesn’t fully express this logic clearly.
- There’s confusion around how this affects `when` clauses and variable naming.
- Using `n` directly in `when:` conditions may lead to misinterpretation of its actual value.

This topic has been discussed in [[PR #530](https://github.com/riscv-software-src/riscv-unified-db/issues#530)](#530).

dhower-qc, 2025-04-09:
To clarify: this is about expressing a transformation to an encoding variable.

The current schema has a `left_shift` operator for variables, but the transformation here is a left shift and an OR.

I see a few ways to resolve this:

1. Add a `left_shift_in_ones` operator, or something similar
    - Keeps information about encoding variable entirely in YAML, without IDL
2. Express transformations using IDL
    - Avoids inevitable situation where we have to keep adding operators
    - For example, this would be `transform: (var << 1) | 1`. `left_shift` would be `transform: var << 1`. Etc.

@drom, I expect you might have an opinion.

ThinkOpenly, 2025-04-09:
> The current schema has a `left_shift` operator for variables, but the transformation here is a left shift and an OR.

> 2. Express transformations using IDL
*>     
>    * For example, this would be `transform: (var << 1) | 1`. `left_shift` would be `transform: var << 1`. Etc.

I suspect we'll eventually need this more general approach.

The Power ISA has an odd one that serves as an example in the "Hash Store" and "Hash Check" instructions:
```
Operand: offset ← EXTS(0b111_1111||DX||D||0b000)
```
- Merge two sub-fields (we do this well already with the flexible "location" attribute)
- Shift left
- Concatenate on the _left_ 0b111_1111
- Sign extend

7908837174, 2025-08-24:
Hi @Shehrozkashif, I'd like to take ownership of Issue #601 regarding instruction field transformations. I've already drafted a related PR that addresses the transformation logic and schema clarity discussed in the thread.

Please assign me to the issue so I can continue refining and aligning the implementation. Happy to iterate based on feedback and ensure it's reviewer-ready.

Thanks!