riscv/riscv-unified-db #610: Add conflicts: Zclsd entry in Zcf.yaml for completeness

Shehrozkashif, 2025-04-11:
Currently, Zclsd.yaml declares a conflict with Zcf, but Zcf.yaml does not declare a reciprocal conflicts: Zclsd. The mutual conflict is technically covered from the Zclsd side, but for completeness and clarity, we should consider whether to explicitly add conflicts: Zclsd in Zcf.yaml as well.

Opening this issue to discuss whether we want to formalize this conflict in both directions.

AFOliveira, 2025-04-13:
I believe @dhower-qc has pronounced in the past to avoid the duplication of such information

Shehrozkashif, 2025-04-13:
I’m not sure about it !

ThinkOpenly, 2025-04-14:
My opinion (expressed in the provoking PR #577) is to just define the conflict in the extension that provoked the conflict. (First-come, first served.) But I can live with a different decision.

AFOliveira, 2025-04-15:
In general, I would say that everyone agrees with not duplicating the information. @Shehrozkashif Duplicating will only making maintaning harder without any true benefit, because we can already gather that information from non-duplicated sources

ThinkOpenly, 2025-04-15:
> In general, I would say that everyone agrees with not duplicating the information. Duplicating will only make maintaning harder without any true benefit, because we can already gather that information from non-duplicated sources

Solely for the purpose of debate, I'll note what you said in #577, @AFOliveira:
> Won't first come first serve be more complicated in the long run? Also, downstream tools will have more problems with dealing with this keyword if they are not duplicated.

Just making sure: are you now content with first-come, first-serve?

ThinkOpenly, 2025-04-15:
To close this, I was going to suggest adding a comment in `schemas/ext_schema.json`, but JSON doesn't permit comments. Then I saw some uses of a `$comment` field, which I presume is for this sort of purpose. I suggest adding a comment such as:
> It is necessary and sufficient to note conflicts between a pair of extensions in just one of those extension definition files. It is not incorrect to note conflicts in both extensions, however the preferred implementation is to only note the conflict in the _later_, intruding extension, as in first-come, first-served.