riscv/riscv-unified-db #612: Wavedrom of amocas.d is not being correctly loaded

AFOliveira, 2025-04-14:
Instruction amocas.d has the following encoding bug. Happens in both PDF/HTML

[image]

kbroch-rivosinc, 2025-04-14:
html ref: https://riscv-software-src.github.io/riscv-unified-db/manual/html/isa/isa_20240411/insts/amocas.d.html

Same problem with https://riscv-software-src.github.io/riscv-unified-db/manual/html/isa/isa_20240411/insts/amocas.q.html

Also didn't we say we were going to get rid of those RV32/RV64 tabs for all inst. since there's so few that have them?

AFOliveira, 2025-04-14:
I believe @dhower-qc had pushed a PR to get rid of them, right?

dhower-qc, 2025-04-16:
I don't think this is an encoding bug. It's just a really long register description that is bleeding into adjacent fields. The register for this instruction must be even. That gets encoded as `not: {1, 3, 5, 7, 9, 11, 13, 15, 17, 19, 21, 23, 25, 27, 29, 31}` and display as `rsN != {1, 3, 5, 7, 9, 11, 13, 15, 17, 19, 21, 23, 25, 27, 29, 31}`. This is probably just a special case we need to detect in the wavedrom generation and display it nicer.

I'll look into why the tabs showed back up.

ThinkOpenly, 2025-07-31:
> I will definitely want to work on this issue but I have a wish kindly if possible could I also give some hand on Zvqdotq extension issue the issue was actually currently assigned to my friend that's why I can't directly comment on that issue but can I raise a PR on that issue? Don't need to assign me, I will try by me on that issue.

I'm not parsing that well. I believe you are referencing PR #902. I'll /cc you there, if that helps. How else can we help?