riscv/riscv-unified-db #622: Assembly is not fully corrected

AFOliveira, 2025-04-16:
Assembly fields was auto-generated out of riscv-opcodes and is mostly wrong. I will check if I can generate this out of LLVM or GCC next week.

ThinkOpenly, 2025-04-16:
The right place to source this information should be upstream (Sail) not downstream, of course, but that's quite a challenge (as I know very well). Further, since the Sail assembly representations aren't really verified or used, I'm not sure how much confidence to have in them. Even further, the "Sail-to-JSON" project of mine (https://github.com/ThinkOpenly/sail) that has now stalled for some time, while it does extract this information from Sail in a more useful format (JSON):
```
  "mnemonic": "add",
  "syntax": "rd,rs1,rs2",
...
```
...that project is also known to have bugs in exactly this area.

I bring this up because it could be a really easy start to validate much of the current content, if not complete, and maybe with some false negatives.

If this is something you want (or want me) to pursue, let me know.

AFOliveira, 2025-04-22:
Paul, this slid past me and I did not read it at the appropriate time. I think LLVM may be a more reliable source; however, it also has its challenges. Do you think we can create a regression for assembly based on Sail? Does Sail specify register type (GPR, FPR...)?

jordancarlin, 2025-04-22:
> Does Sail specify register type (GPR, FPR...)?

The sail model uses separate types for the different kinds of registers (this is a relatively recent change). This ends up looking like the following:

```
union clause ast = MOVETYPEX : (regidx, vregidx)

mapping clause assembly = MOVETYPEX(rs1, vd)
  <-> "vmv.v.x" ^ spc() ^ vreg_name(vd) ^ sep() ^ reg_name(rs1)
```

AFOliveira, 2025-04-22:
> The sail model uses separate types for the different kinds of registers (this is a relatively recent change). This ends up looking like the following:
>
> ```
> union clause ast = MOVETYPEX : (regidx, [...]

Thanks for the answer Jordan! However, without Paul's change there is no way of generating a json(or any easily parseable format) out of Sail that has all the instructions names and arguments, right?

jordancarlin, 2025-04-22:
> Thanks for the answer Jordan! However, without Paul's change there is no way of generating a json(or any easily parseable format) out of Sail that has all the instructions names and arguments, right?

You might be able to use the documentation json that Sail can produce. It's primarily intended for including Sail snippets in documentation (like the asciidoc manuals), but depending on how you are parsing the file it might work for you. Here's a version compiled from the latest Sail model if you want to take a look: [riscv_model_rv64d.json](https://github.com/user-attachments/files/19845844/riscv_model_rv64d.json).

The relevant portion would be the assembly mappings, which look like this:
```
"source": "mapping clause assembly = MOVETYPEX(rs1, vd)\n  <-> \"vmv.v.x\" ^ spc() ^ vreg_name(vd) ^ sep() ^ reg_name(rs1)",
          "left": {
            "type": "app",
            "id": "MOVETYPEX",
            "patterns": [
              { "type": "id", "id": "rs1" },
              { "type": "id", "id": "vd" }
            ]
          },
          "right": {
            "type": "string_append",
            "patterns": [
              { "type": "literal", "value": "\"vmv.v.x\"" },
              {
                "type": "app",
                "id": "spc",
                "patterns": [{ "type": "literal", "value": "()" }]
              },
              {
                "type": "app",
                "id": "vreg_name",
                "patterns": [{ "type": "id", "id": "vd" }]
              },
              {
                "type": "app",
                "id": "sep",
                "patterns": [{ "type": "literal", "value": "()" }]
              },
              {
                "type": "app",
                "id": "reg_name",
                "patterns": [{ "type": "id", "id": "rs1" }]
              }
            ]
          }
        },
```

ThinkOpenly, 2025-05-19:
> You might be able to use the documentation json that Sail can produce. It's primarily intended for including Sail snippets in documentation (like the asciidoc manuals), but depending on how you are [...]

Very interesting. @jordancarlin, How does one produce that JSON from the Sail model source? I see that things have migrated to `cmake`, so now I'm unable to figure anything out. ;-)

ThinkOpenly, 2025-05-19:
Found it. https://github.com/riscv/sail-riscv/blob/master/.github/workflows/release.yml#L69-L71:
```
          cmake -S . -B build -GNinja -DCMAKE_BUILD_TYPE=Release -DSTATIC=TRUE
          cd build
          ninja all generated_docs_rv64d
```

jordancarlin, 2025-05-20:
Yup. That should work. If you just want the documentation json then drop the `all` from the ninja line. If you don't want to use ninja you can also do
```
cmake -S . -B build -DCMAKE_BUILD_TYPE=Release
cmake --build build --target generated_docs_rv64d
```