riscv/riscv-unified-db #639: Not all artifact links work after pages.yaml calls deploy.sh

james-ball-qualcomm, 2025-04-21:
Was checking out the links in https://riscv-software-src.github.io/riscv-unified-db/index.html after the PR that merged my CSC changes. Only some of the artifact links work. I looked at the output log and the deploys.sh and for the life of me can't figure out why some of the artifacts are missing since they all seem to be built and copied. 

My recommendation is that the deploy pages checks the exit status of the calls to "do" to build each artifact and also the exit status of the "cp" command used to copy each artifact into the _site directory. Right now without checking the exit status the deploy.sh script could pass even though it couldn't build/copy each required artifact.

kbroch-rivosinc, 2025-04-21:
looks like 10 missing (404) links: 

```
❯ linkchecker -r 1 https://riscv-software-src.github.io/riscv-unified-db/index.html
INFO linkcheck.cmdline 2025-04-21 07:56:35,542 MainThread Checking intern URLs only; use --check-extern to check extern URLs.
LinkChecker 10.5.0
Copyright (C) 2000-2016 Bastian Kleineidam, 2010-2024 LinkChecker Authors
LinkChecker comes with ABSOLUTELY NO WARRANTY!
This is free software, and you are welcome to redistribute it under
certain conditions. Look at the file `COPYING' within this distribution.
Read the documentation at https://linkchecker.github.io/linkchecker/
Write comments and bugs to https://github.com/linkchecker/linkchecker/issues

Start checking at 2025-04-21 07:56:35-007
10 threads active,    12 links queued,    4 links in  26 URLs checked, runtime 1 seconds

URL        `https://riscv-software-src.github.io/riscv-unified-db/pdfs/RVI20.pdf'
Name       `RVI20 Profile Release'
Parent URL https://riscv-software-src.github.io/riscv-unified-db/index.html, line 37, col 11
Real URL   https://riscv-software-src.github.io/riscv-unified-db/pdfs/RVI20.pdf
Check time 1.923 seconds
Size       5KB
Result     Error: 404 Not Found

URL        `https://riscv-software-src.github.io/riscv-unified-db/isa_explorer.xlsx'
Name       `Excel version (includes Extensions, Instructions, CSRs)'
Parent URL https://riscv-software-src.github.io/riscv-unified-db/index.html, line 31, col 11
Real URL   https://riscv-software-src.github.io/riscv-unified-db/isa_explorer.xlsx
Check time 2.615 seconds
Size       5KB
Result     Error: 404 Not Found

URL        `https://riscv-software-src.github.io/riscv-unified-db/pdfs/RVA20.pdf'
Name       `RVA20 Profile Release'
Parent URL https://riscv-software-src.github.io/riscv-unified-db/index.html, line 38, col 11
Real URL   https://riscv-software-src.github.io/riscv-unified-db/pdfs/RVA20.pdf
Check time 3.043 seconds
Size       5KB
Result     Error: 404 Not Found

URL        `https://riscv-software-src.github.io/riscv-unified-db/pdfs/RVA22.pdf'
Name       `RVA22 Profile Release'
Parent URL https://riscv-software-src.github.io/riscv-unified-db/index.html, line 39, col 11
Real URL   https://riscv-software-src.github.io/riscv-unified-db/pdfs/RVA22.pdf
Check time 3.355 seconds
Size       5KB
Result     Error: 404 Not Found

URL        `https://riscv-software-src.github.io/riscv-unified-db/pdfs/RVA23.pdf'
Name       `RVA23 Profile Release'
Parent URL https://riscv-software-src.github.io/riscv-unified-db/index.html, line 40, col 11
Real URL   https://riscv-software-src.github.io/riscv-unified-db/pdfs/RVA23.pdf
Check time 3.876 seconds
Size       5KB
Result     Error: 404 Not Found

URL        `https://riscv-software-src.github.io/riscv-unified-db/pdfs/RVB23.pdf'
Name       `RVB23 Profile Release'
Parent URL https://riscv-software-src.github.io/riscv-unified-db/index.html, line 41, col 11
Real URL   https://riscv-software-src.github.io/riscv-unified-db/pdfs/RVB23.pdf
Check time 3.844 seconds
Size       5KB
Result     Error: 404 Not Found

URL        `https://riscv-software-src.github.io/riscv-unified-db/pdfs/AC100-CRD.pdf'
Name       `AC100 CRD (based on RVB23)'
Parent URL https://riscv-software-src.github.io/riscv-unified-db/index.html, line 47, col 11
Real URL   https://riscv-software-src.github.io/riscv-unified-db/pdfs/AC100-CRD.pdf
Check time 3.955 seconds
Size       5KB
Result     Error: 404 Not Found

URL        `https://riscv-software-src.github.io/riscv-unified-db/pdfs/AC200-CRD.pdf'
Name       `AC200 CRD (based on RVA23)'
Parent URL https://riscv-software-src.github.io/riscv-unified-db/index.html, line 48, col 11
Real URL   https://riscv-software-src.github.io/riscv-unified-db/pdfs/AC200-CRD.pdf
Check time 3.592 seconds
Size       5KB
Result     Error: 404 Not Found
 9 threads active,     0 links queued,   17 links in  26 URLs checked, runtime 6 seconds

URL        `https://riscv-software-src.github.io/riscv-unified-db/ruby/idl/index.html'
Name       `IDL language documentation'
Parent URL https://riscv-software-src.github.io/riscv-unified-db/index.html, line 73, col 11
Real URL   https://riscv-software-src.github.io/riscv-unified-db/ruby/idl/index.html
Check time 3.860 seconds
Size       5KB
Result     Error: 404 Not Found

URL        `https://riscv-software-src.github.io/riscv-unified-db/ruby/arch_def/index.html'
Name       `Ruby UDB interface documentation'
Parent URL https://riscv-software-src.github.io/riscv-unified-db/index.html, line 74, col 11
Real URL   https://riscv-software-src.github.io/riscv-unified-db/ruby/arch_def/index.html
Check time 3.657 seconds
Size       5KB
Result     Error: 404 Not Found

Statistics:
Downloaded: 4KB.
Content types: 0 image, 17 text, 0 video, 0 audio, 8 application, 0 mail and 1 other.
URL lengths: min=64, max=103, avg=76.

That's it. 26 links in 26 URLs checked. 0 warnings found. 10 errors found.
Stopped checking at 2025-04-21 07:56:45-007 (10 seconds)
```

james-ball-qualcomm, 2025-04-21:
I'm contemplating adding the proposed better error checking to deploy.sh today. I can run it on my local workspace but it already works fine there so I'm need to actually run it using the GitHub pages.yml as a GitHub Action without getting a PR approved and merged into main? I ask because Derek is out-of-town this week and I'm not sure any other contributor has permissions to review and approve PRs so they get merged.

@ThinkOpenly or @kbroch-rivosinc, do you have such permissions?

Also, could I add a call to "linkchecker -r 1 https://riscv-software-src.github.io/riscv-unified-db/index.html" at the end of the pages.yml or deploy.sh to make sure what just got copied to the github.io site doesn't have broken links? If there were broken links, could I have the action fail and not update the github.io? Seems like it would be too late.

dhower-qc, 2025-04-25:
We can add the deploy workflow as a PR requirement so that things like this would be checked before going to main. However, given the current run time of deploy.sh, I think that'd be counterproductive.

AFOliveira, 2025-04-30:
> We can add the deploy workflow as a PR requirement so that things like this would be checked before going to main. However, given the current run time of deploy.sh, I think that'd be counterproductive.

is there a way to allow a branch that is not main to deploy to a different URL?

dhower-qc, 2025-04-30:
> > We can add the deploy workflow as a PR requirement so that things like this would be checked before going to main. However, given the current run time of deploy.sh, I think that'd be counterproductive.
> 
> is there a way to allow a branch that is not main to deploy to a different URL?

Sure. Right now, the deployment is just running a shell script. So, we could get the URL from, say, an environment variable.

ThinkOpenly, 2025-04-30:
> Sure. Right now, the deployment is just running a shell script. So, we could get the URL from, say, an environment variable.

`lib/deploy.sh`?
```
PAGES_URL="https://riscv-software-src.github.io/riscv-unified-db"
```

What are the restrictions for a random person to deploy a random branch?

AFOliveira, 2025-04-30:
Maybe anyone with right access would be restrict enough, right? I think we may also branch-protect a branch that is not main, so it would only go through after an approving review

ThinkOpenly, 2025-04-30:
> ```
> PAGES_URL="https://riscv-software-src.github.io/riscv-unified-db"
> ```
> 
> What are the restrictions for a random person to deploy a random branch?

I was actually asking just about what URLs can be used. (Sorry for being unclear.) I want to make sure the canonical site cannot be overwritten, and that users have some idea about what URL to specify. Do they need to specify their own github.io space (user.github.io/anything)?

dhower-qc, 2025-04-30:
Now that I think about this more, the URL is actually dictated by GitHub -- we are just capturing what's going to happen.

The deploy script writes the files we want to host into a local `_site` directory, and then `pages.yml` workflow uploads it:

https://github.com/riscv-software-src/riscv-unified-db/blob/cf88b2bef72f5f517da111810a384e9021a67833/.github/workflows/pages.yml#L27-L34

AFOliveira, 2025-05-02:
So, every deployment (test or not test) will overwrite the previous one? If so, I think we will have to either use [ACT ](https://github.com/nektos/act) to test locally or use another way of testing. Can't we test `deploy.sh` in yet another test-suite that would be need to be run by the server before the merge-queue would write to main?

ThinkOpenly, 2025-05-02:
I imagine there's some way to customize the URL that github-pages ends up using, but I'm not an expert.

In lieu of that, might we be content to generate the content in the "_site" directory, and leave it there for inspection? It's easy enough to launch a browser at a local file (presuming the embedded links are relative)?

dhower-qc, 2025-05-05:
The deployment runtime is the only issue for checking this either locally or as part of PR CI. A few options:

1. Hidden in #708 is a change to `deploy.sh` that locally parallelizes part of the run. (yes, @ThinkOpenly, I should make that separate PR 😄)
2. We can use ACT as @AFOliveira suggests. This would let us parallelize the deployment like we do with `regress.yml` and still run locally. The issue here is that ACT requires Docker (which is reasonable, since that's what GitHub is using), but causes some issues for Docker-averse companies like mine or users without sudo.
3. I think that we can have a different set of tests run from the merge queue than what we have on a PR update. That way we could run the `regress.yml` tests as we do now, but run regress+deploy in the merge queue. Maybe combined with (1) that's good enough?

closed 2025-05-20 by james-ball-qualcomm

kbroch-rivosinc, 2025-05-20:
Rest of this missing links covered by this: https://github.com/riscv-software-src/riscv-unified-db/issues/769