riscv/riscv-unified-db #642: check-jsonschema pre-commit hook doesn't validate arch_overlay dir files

kbroch-rivosinc, 2025-04-21:
Currently only arch dir yaml files are being compared to check-jsonschema pre-commit hook
arch_overlay has other yaml files that should but those problems are flagged to CI ruby code here: https://github.com/riscv-software-src/riscv-unified-db/actions/runs/14578986964/job/40891298108?pr=637#step:4:78

kbroch-rivosinc, 2025-04-21:
I'll put a PR in for this.

kbroch-rivosinc, 2025-04-21:
Actually this might not work if the overlay is only meant to override some part of an existing arch yaml files like marchid then trying to compare it against the jsonschema files will result in problems like this: 

```
Validate files with jsonschema...........................................Failed
- hook id: check-jsonschema
- exit code: 1

ok -- validation done
ok -- validation done
ok -- validation done
ok -- validation done
Schema validation errors were encountered.
  arch_overlay/example/csr/marchid.yaml::$: '$schema' is a required property
  arch_overlay/example/csr/marchid.yaml::$: 'kind' is a required property
  arch_overlay/example/csr/marchid.yaml::$: 'name' is a required property
  arch_overlay/example/csr/marchid.yaml::$: 'long_name' is a required property
  arch_overlay/example/csr/marchid.yaml::$: 'length' is a required property
  arch_overlay/example/csr/marchid.yaml::$: 'description' is a required property
  arch_overlay/example/csr/marchid.yaml::$: 'priv_mode' is a required property
  arch_overlay/example/csr/marchid.yaml::$: 'definedBy' is a required property
  arch_overlay/example/csr/marchid.yaml::$.fields.Architecture: {'description': 'Custom overlay description'} is not valid under any of the given schemas
  Underlying errors caused this.

  Best Match:
    $.fields.Architecture: 'location' is a required property

  2 other errors were produced. Use '--verbose' to see all errors.
  arch_overlay/example/csr/marchid.yaml::$.fields.Architecture: {'description': 'Custom overlay description'} is not valid under any of the given schemas
  Underlying errors caused this.

  Best Match:
    $.fields.Architecture: 'type' is a required property

  1 other errors were produced. Use '--verbose' to see all errors.
  arch_overlay/example/csr/marchid.yaml::$.fields.Architecture: {'description': 'Custom overlay description'} is not valid under any of the given schemas
  Underlying errors caused this.

  Best Match:
    $.fields.Architecture: 'reset_value' is a required property

  1 other errors were produced. Use '--verbose' to see all errors.
  arch_overlay/example/csr/marchid.yaml::$: 'virtual_address' is a required property
  arch_overlay/example/csr/marchid.yaml::$: {'fields': {'Architecture': {'description': 'Custom overlay description'}}} is not valid under any of the given schemas
  Underlying errors caused this.

  Best Match:
    $: 'address' is a required property

  1 other errors were produced. Use '--verbose' to see all errors.
ok -- validation done
ok -- validation done
ok -- validation done
ok -- validation done
ok -- validation done
ok -- validation done
ok -- validation done

```

dhower-qc, 2025-04-25:
Yea, we shouldn't expect arch_overlay files to validate against anything; only the merge result should validate (and that is checked via ruby/python through the do flows)

kbroch-rivosinc, 2025-04-25:
> Yea, we shouldn't expect arch_overlay files to validate against anything; only the merge result should validate (and that is checked via ruby/python through the do flows)

Thanks for confirming.  I'll close this as there's nothing to do here.

closed 2025-04-25 by kbroch-rivosinc