riscv/riscv-unified-db #645: Error in reuse lint

AFOliveira, 2025-04-22:
Locally, the reuse lint pre-commit hook is throwing errors for files that I haven't even commited. This is quite odd, since those files sometimes are helper scripts and/or cache files which I don't want to be verified, but still want to keep them in my local repo. Is this a "me" problem, or is there anything misconfigured on the reuse lint?

## Example 

I am adding only arch/inst/V but getting errors for:

```
reuse lint...............................................................Failed
- hook id: reuse
- exit code: 1

# MISSING COPYRIGHT AND LICENSING INFORMATION

The following files have no copyright and licensing information:
* et --hard 1ce9929d2994a3bd283fb568bb4ffc4f91e9ecd1
* ext/auto-inst/__pycache__/parsing.cpython-312.pyc
* ext/auto-inst/__pycache__/test_parsing.cpython-312-pytest-8.3.4.pyc
* ext/llvm-project/llvm.py
* ext/llvm-project/riscv.json
* inst.go
* processed_instr_dict.json

# SUMMARY

* Bad licenses: 0
* Deprecated licenses: 0
* Licenses without file extension: 0
* Missing licenses: 0
* Unused licenses: 0
* Used licenses: BSD-3-Clause-Clear, MIT, CC-BY-4.0, BSD-2-Clause
* Read errors: 0
* Files with copyright information: 2345 / 2352
* Files with license information: 2345 / 2352

Unfortunately, your project is not compliant with version 3.3 of the REUSE Specification :-(
```

AFOliveira, 2025-04-22:
cc @kbroch-rivosinc

kbroch-rivosinc, 2025-04-23:
reuse ignores https://reuse.software/spec-3.3/#covered-and-ignored-files
If you think it's behaving badly still let me know what PR to take a look at.

AFOliveira, 2025-04-23:
@kbroch-rivosinc maybe I did not explain myself correctly, but what always made sense to me is that pre-commit hooks check things that are being commited. In this case, the code that is throwing errors is not part of the commit, it is in the unommited changes, so I find this quite odd and can't figure why it is being flagged. My issue was to try to figure if we have some misconfiguration that is looking in the whole repo instead of just the commit we are submitting

kbroch-rivosinc, 2025-04-24:
I had set it up to `reuse lint` the whole project https://github.com/fsfe/reuse-tool/blob/main/.pre-commit-hooks.yaml
but I guess that's problematic if there's files that won't be committed and also shouldn't be `.gitignore`d.

There is a file-only hook we can deploy instead that behaves as you expected.

PR #658 (2025-04-28): fix: change reuse lint pre-commit hook to only look at committed files
fixes #645
previously looked at all files and didn't consider the case of dirty files that won't be committed or gitignored which is a common case
commit 9ff372ed (2025-04-28): fix: change reuse lint pre-commit hook to only look at committed files (#658)

fixes #645
previously looked at all files and didn't consider the case of dirty
files that won't be committed or gitignored which is a common case
closed 2025-04-28 by kbroch-rivosinc