riscv/riscv-unified-db #646: Antora HTML Template has useless buttons

AFOliveira, 2025-04-22:
I realized there are three dead buttons on our html template. Is this something that antora forces us to use, or is it modifiable? I think we do not want those lying there forever.

[image]

AFOliveira, 2025-04-22:
cc @dhower-qc @ddrcode 

Maybe this is something @ddrcode knows better than all of us :)

kbroch-rivosinc, 2025-04-23:
Here's how this is happening currently: 

This https://github.com/riscv-software-src/riscv-unified-db/blob/main/backends/manual/templates/playbook.yml.erb#L23

```
ui:
  bundle:
    url: https://gitlab.com/antora/antora-ui-default/-/jobs/artifacts/HEAD/raw/build/ui-bundle.zip?job=bundle-stable
```

pulls in this: 

```html
❯ cat partials/header-content.hbs
<header class="header">
  <nav class="navbar">
    <div class="navbar-brand">
      <a class="navbar-item" href="{{{or site.url siteRootPath}}}">{{site.title}}</a>
      {{#if env.SITE_SEARCH_PROVIDER}}
      <div class="navbar-item search hide-for-print">
        <div id="search-field" class="field">
          <input id="search-input" type="text" placeholder="Search the docs"{{#if page.home}} autofocus{{/if}}>
        </div>
      </div>
      {{/if}}
      <button class="navbar-burger" aria-controls="topbar-nav" aria-expanded="false" aria-label="Toggle main menu">
        <span></span>
        <span></span>
        <span></span>
      </button>
    </div>
    <div id="topbar-nav" class="navbar-menu">
      <div class="navbar-end">
        <a class="navbar-item" href="#">Home</a>
        <div class="navbar-item has-dropdown is-hoverable">
          <a class="navbar-link" href="#">Products</a>
          <div class="navbar-dropdown">
            <a class="navbar-item" href="#">Product A</a>
            <a class="navbar-item" href="#">Product B</a>
            <a class="navbar-item" href="#">Product C</a>
          </div>
        </div>
        <div class="navbar-item has-dropdown is-hoverable">
          <a class="navbar-link" href="#">Services</a>
          <div class="navbar-dropdown">
            <a class="navbar-item" href="#">Service A</a>
            <a class="navbar-item" href="#">Service B</a>
            <a class="navbar-item" href="#">Service C</a>
          </div>
        </div>
        <div class="navbar-item">
          <span class="control">
            <a class="button is-primary" href="#">Download</a>
          </span>
        </div>
      </div>
    </div>
  </nav>
</header>
```

I believe I can just add an override in the `playbook.yml.erb` file to remove these.  PR to follow.

AFOliveira, 2025-04-23:
Awesome, thanks @kbroch-rivosinc!

closed 2025-04-28 by kbroch-rivosinc