riscv/riscv-unified-db #647: Zaamo Assembly

AFOliveira, 2025-04-22:
Zaamo instructions have parenthesis around xs1, is there any reason for this?

```
name: amoadd.d
long_name: Atomic fetch-and-add doubleword
description: |
  Atomically:

    * Load the doubleword at address _rs1_
    * Write the loaded value into _rd_
    * Add the value of register _rs2_ to the loaded value
    * Write the sum to the address in _rs1_
definedBy: Zaamo
base: 64
assembly: xd, xs2, (xs1)

```

ThinkOpenly, 2025-04-22:
> Zaamo instructions have parenthesis around xs1, is there any reason for this?

Are you asking why they're like this in UDB, or in the spec? That's how they are in Sail as well:
```
mapping clause assembly = AMO(op, aq, rl, rs2, rs1, width, rd)
  <-> amo_mnemonic(op) ^ "." ^ size_mnemonic(width) ^ maybe_aq(aq) ^ maybe_rl(rl) ^ spc() ^ reg_name(rd) ^ sep() ^ reg_name(rs2) ^ sep() ^ "(" ^ reg_name(rs1) ^ ")"
```

That expression indicates that the value being operated on by the instruction is the not the value in the register, but the value _at the memory address_ in the register.

AFOliveira, 2025-04-22:
@ThinkOpenly Oh, I probably should ahve looked up the spec instead of just looking up who commited it. But does it make sense from and assembly perspective? LLVM does not return anything with parenthesis so I wonder if that information is something that should be captured in instruction semantics rather than in the assembly field

ThinkOpenly, 2025-04-22:
> does it make sense from and assembly perspective?

I'm not a fan, but there is 50+ years of history to fight against. ;-)

> LLVM does not return anything with parenthesis

meaning?

> so I wonder if that information is something that should be captured in instruction semantics rather than in the assembly field

Not following this, either. "instruction semantics" = `operation()`? What do you mean by "captured"?

For example/discussion, from `lw.yaml`:
```
assembly: xd, imm(xs1)
operation(): |
  XReg virtual_address = X[xs1] + $signed(imm);

  X[xd] = $signed(read_memory<32>(virtual_address, $encoding));
```

The value in `xs1` is read, added to `imm`, then _that_ result is used to access memory.

AFOliveira, 2025-04-22:
> assembly: xd, imm(xs1)

For some reason, this makes much more sense than just `(rs1)` , maybe it's just because I have been exposed to one and not to the other. I'll see how LLVM represents the register actually meaning the address with the reg value to see if I can automate this tiny corner case. Thanks for explaining!

dhower-qc, 2025-04-28:
ok to close?

closed 2025-04-28 by AFOliveira