riscv/riscv-unified-db #648: Differentiate between arguments

AFOliveira, 2025-04-22:
In instructions, there are variables that are assembly defined - like the normal regs - rs1,rs2,rd - and other variables that actually cause a different instructions - i.e. aq, rl. We need to differentiate them in order to generate all valid instructions out of UDB.

### Example 

amoadd.b.yaml:
```
name: amoadd.b
long_name: No synopsis available.
description: |
  No description available.
definedBy: Zabha
assembly: xd, xs2, xs1
encoding:
  match: 00000------------000-----0101111
  variables:
    - name: aq
      location: 26-26
    - name: rl
      location: 25-25
    - name: rs2
      location: 24-20
    - name: rs1
      location: 19-15
    - name: rd
      location: 11-7
```

This will have an `assembly: xd,xs1,xs2`, but it should turn into 4 instructions:

-  amoadd.b
- amoadd.b.aq
- amoadd.b.rl
- amoadd.b.aq.rl

AFOliveira, 2025-04-22:
I know we created separate files for all Zcmop instructions, which I believe fall in the same category, but do we really want to create several files for each of this, or is it just better if we define a field in those variables that specifys this behaviour?

Because, I am afraid of the amount of instruction vector may have that also follow this pattern.

ThinkOpenly, 2025-04-22:
We already have 252 `v*seg*`, what's a few thousand more? ;-)

I worry about making it easy for downstream users. It should be easy to, say, list all instruction mnemonics. If whatever we choose to do allows that, I can support it. :-)

Maybe this overlaps, at least a bit, with #649? ...tooling that allows users to easily interact with the UDB content. (A UDB CLI?)

AFOliveira, 2025-04-22:
> We already have 252 `v*seg*`, what's a few thousand more? ;-)
>
> I worry about making it easy for downstream users. It should be easy to, say, list all instruction mnemonics. If whatever we choose to [...]

Yeah, I agree with your view, plain information ends up by just by being easier. We did otherwise in Mop.R.N and I think we also will need to undo that change to be conforming. Just the other day I found somewhere in the Ruby code the "decoder" for transforming that instruction in the several that it describes. I think if @dhower-qc also agrees, we may even state that in the UDB documentation in #571