riscv/riscv-unified-db #649: Helper/Migration scripts

AFOliveira, 2025-04-22:
We need to provide tools so people migrate from other repos (like riscv-opcodes) into UDB in a seamless way. I have done already done plenty of this work for instructions, both from RISC-V opcodes but also from LLVM. However the scripts I used were never merged because they were one-time use only. However, this is not true anymore, since people may use them in downstream repos. I suggest we create a /helpers or /backends/helpers folder where we can store them, but I want to hear if there are any other thoughts on this.

dhower-qc, 2025-04-28:
In line with the ongoing discussion about repository structure / splitting, I'm thinking we might want to do a big top-down reorganization. Something like

```
bin
data
  arch
  arch_overlay
  cfgs
  schemas
doc
tools
  gems
    idl (IDL compiler)
    udb (Ruby UDB interface)
    ...
  generators
    isa_manual
    libhart
    encoding_h
    ...
  migrators       <--- Helper/migrations scripts under here
    from_opcodes
    ...
```

AFOliveira, 2025-04-29:
Agree

dhower-qc, 2025-04-29:
@ThinkOpenly, thoughts?

ThinkOpenly, 2025-04-29:
Where would `lib` go, or does it persist as-is?
Where do tests go?

Given the disruption, is this an opportunity to consider UDB as being able to host other arches? (Likely getting far ahead of things.) This would maybe require a "riscv" subdirectory under "arch" and moving "arch_overlay" to "arch/riscv/overlay" ? And "cfgs" to arch/riscv/cfgs"?

AFOliveira, 2025-04-29:
> Given the disruption, is this an opportunity to consider UDB as being able to host other arches? (Likely getting far ahead of things.) This would maybe require a "riscv" subdirectory under "arch" and moving "arch_overlay" to "arch/riscv/overlay" ? And "cfgs" to arch/riscv/cfgs"?

Is there any particular reason for us to do so? It's not like UDB is established as the de-facto standard for RISC-V, so I wonder if that isn't a huge stretch, at least for now

dhower-qc, 2025-04-29:
> Where would lib go, or does it persist as-is?

Lib goes away, replaced by proper gems (or, for yaml_resolver.py, a python package).

> Where do tests go?

For gems, there is a standard folder structure (https://guides.rubygems.org/what-is-a-gem/), so their tests will go under gems/NAME/test. I think there is a similar structure for python packages.

We could do something similar for the generators, under generators/NAME/test.

The top-level Rakefile can still discover all the tests and run them.

dhower-qc, 2025-04-29:
> Is there any particular reason for us to do so? It's not like UDB is established as the de-facto standard for RISC-V, so I wonder if that isn't a huge stretch, at least for now

I like your ambition ;) I see no harm in adding an extra directory layer, so we might as well.