riscv/riscv-unified-db #650: Validate proper indentation

ThinkOpenly, 2025-04-22:
**Is your feature request related to a problem? Please describe.**
Improper indentation should be detected before merging.

**Describe the solution you'd like**
I'd like to see tests added that validate indentation within data files and the fields therein:
- YAML (may already be done?)
  - IDL (not checked... would a SystemVerilog formatter work?)
- JSON (not checked... I'm sure there are JSON formatters out there)

kbroch-rivosinc, 2025-04-22:
in general, YAML and JSON are formatted by prettier pre-commit https://github.com/riscv-software-src/riscv-unified-db/blob/main/.pre-commit-config.yaml#L29

default config: 

```
kbroch@kbroch-mac.local:~/rvi/non-repo-repos/riscv-unified-db on  main via  v22.14.0 via 🐍 v3.13.3 via 💎 v3.2.3
❯ cat .prettierrc
{}
```

some files being ignored: 

```
kbroch@kbroch-mac.local:~/rvi/non-repo-repos/riscv-unified-db on  main via  v22.14.0 via 🐍 v3.13.3 via 💎 v3.2.3
❯ cat .prettierignore
gen
.home
.singularity
node_modules
.asciidoctor
.bundle
.git
.github
.vscode
docs/ruby
ext
```

AFOliveira, 2025-04-28:
@kbroch-rivosinc does IDL gets indented with this? Maybe that would be a big effort, but would it be possible to extend prettier to do it?  I think most of the indentation @ThinkOpenly is refering to is IDL related.

dhower-qc, 2025-04-28:
There isn't a formatter for IDL. Looking briefly at what it would take, I see that the [Ruby plugin](https://github.com/prettier/plugin-ruby/blob/main/src/server.rb) for prettier actually just wraps a call to a formatter implemented in Ruby via javascript. That seems like the easiest path here, too. If we add a formatted output function to the existing IDL syntax tree, we can just call it from javascript.

FYI, there is a `to_idl` function on the IDL AST, but it prints everything in minimal format. We'd just need to add a similar function that adds nice whitespace.

syedowaisalishah, 2025-05-08:
@dhower-qc @ThinkOpenly
Hi! I'd like to work on the IDL formatting part of this issue by adding a pretty-printer to the IDL AST in Ruby and integrating it with pre-commit hooks. Please let me know if anyone is already working on this or if you have any suggestions or advice before I begin. Thanks!

ThinkOpenly, 2025-05-08:
> Please let me know if anyone is already working on this or if you have any suggestions or advice before I begin.

There are suggestions above, of course. @kbroch-rivosinc has looked into this more than anyone I'm aware of. OK to assign this to @syedowaisalishah , @kbroch-rivosinc ?

dhower-qc, 2025-05-08:
One thing that is going to be tricky here, assuming we go the route of using the IDL AST to pretty print:

**The AST doesn't capture comments or whitespace**. So, the "pretty printed" text will be nicely indented, but uncommented and "squished" code.

A couple options:

1. We could start capturing comments/whitespace in the AST. Even then, we have to decide how to display them when formatting.
2. We go a different route with the pretty printer. If all we want to do is make sure that scopes are indented properly, that's more tractable than trying to format the entire language.

My very slight preference would be option 1., since that would at least set us up for more complex formatting down the road. But it could also turn into more work than we want to take on right now, which would change my preference to 2.