riscv/riscv-unified-db #678: Cpp Hart Regression test is always failing

AFOliveira, 2025-04-28:
Initial pointed by @james-ball-qualcomm - the regression test related to the cpp_hart is failing for all PRs with the following log:
Relates with #670 
```
/workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task_manager.rb:59:in `[]': Don't know how to build task '/workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/cfgs/rv64/inst.hxx.unformatted' (See the list of available tasks with `do --tasks`) (RuntimeError)
Did you mean?  /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/bitfield.hxx.unformatted
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/hart_factory.hxx.unformatted
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/enum.hxx.unformatted
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/db_data.hxx.unformatted
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/libhart.h.unformatted
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/inst.hpp
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/defines.hpp
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/src/enum.cxx.unformatted
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/bitfield.hpp
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/bits.hpp
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/src/db_data.cxx.unformatted
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/config_validator.hpp
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/renode_imports.h
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/util.hpp
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/stop_reason.h
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/xregister.hpp
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/renode_map.h
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/version.hpp
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/soc_model.hpp
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/csr.hpp
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/iss_soc_model.hpp
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/bits-yaml.hpp
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/cpp_exceptions.hpp
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/memory.hpp
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/hart.hpp
               /workspaces/riscv-unified-db/gen/cpp_hart_gen/rv64_RelWithDebInfo/include/udb/elf_reader.hpp
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:405:in `[]'
	from /workspaces/riscv-unified-db/backends/cpp_hart_gen/tasks.rake:312:in `block (3 levels) in <top (required)>'
	from /workspaces/riscv-unified-db/backends/cpp_hart_gen/tasks.rake:310:in `each'
	from /workspaces/riscv-unified-db/backends/cpp_hart_gen/tasks.rake:310:in `block (2 levels) in <top (required)>'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:281:in `block in execute'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:281:in `each'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:281:in `execute'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:219:in `block in invoke_with_call_chain'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:199:in `synchronize'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:199:in `invoke_with_call_chain'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:243:in `block in invoke_prerequisites'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:241:in `each'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:241:in `invoke_prerequisites'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:218:in `block in invoke_with_call_chain'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:199:in `synchronize'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:199:in `invoke_with_call_chain'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task.rb:188:in `invoke'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:188:in `invoke_task'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `block (2 levels) in top_level'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `each'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `block in top_level'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:147:in `run_with_threads'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:132:in `top_level'
	from -e:1:in `<main>'
Running with 1 job(s)
```

dhower-qc, 2025-04-28:
These are bugs introduced by #584. Fix coming shortly.

james-ball-qualcomm, 2025-04-29:
Thanks for the fix @dhower-qc!

commit 73838c89 (2025-04-29): fix(cpp): re-add functions deleted by #584

fixes #678
commit 1d53a7c9 (2025-04-29): fix(cpp): re-add functions deleted by #584 (#681)

fixes #678
PR #681 (2025-04-29): fix(cpp): re-add functions deleted by #584
fixes #678
commit 2ede9a34 (2025-04-29): fix(cpp): re-add functions deleted by #584 (#681)

fixes #678
closed 2025-04-29 by james-ball-qualcomm