riscv/riscv-unified-db #696: feat: commitlint for PR titles / descriptions

dhower-qc, 2025-04-29:
We are currently using the policy that all PRs are squash-merged. The PR title becomes the squash commit title and the PR description becomes the commit message.

We have commitlint running using pre-commit, but that's not checking what actually shows up in main history since it doesn't run on the PR.

dhower-qc, 2025-04-29:
@kbroch-rivosinc, do you know if this is possible?

kbroch-rivosinc, 2025-05-01:
Here's short community discussion on this topic: https://github.com/orgs/community/discussions/8688

I'm wondering if we can run a GH action once in the [merge queue](https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/configuring-pull-request-merges/managing-a-merge-queue#configuring-continuous-integration-ci-workflows-for-merge-queues) that runs `pre-commit run commitlint` and uses the `--to-ref` and `--from-ref` https://pre-commit.com/#pre-commit-run

Shehrozkashif, 2025-05-17:
Would it be possible to validate the PR title and description format using commitlint directly in the merge queue or before squash merging, so we ensure that only valid titles/descriptions make it into the commit history?

Also, is there a recommended way to block a PR from being merged if the title or description doesn’t pass commitlint rules  perhaps using a required status check from a GitHub Action?

kbroch-rivosinc, 2025-05-17:
There's a github action event called [`merge_group`](https://docs.github.com/en/actions/writing-workflows/choosing-when-your-workflow-runs/events-that-trigger-workflows#merge_group) which we have enabled [here](https://github.com/riscv-software-src/riscv-unified-db/blob/main/.github/workflows/regress.yml#L9).

I would think as long as ["commit squashing"](https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/configuring-pull-request-merges/configuring-commit-squashing-for-pull-requests) happens before the PR enters the [merge queue](https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/configuring-pull-request-merges/managing-a-merge-queue), that the `merge_group` event would trigger `pre-commit` to run.  I think the problem might be what I mentioned above that the `pre-commit` doesn't think it has any work to do because the default commits to compare across aren't obvious in the merge queue.

Getting this to work would be the recommended way.

Short of that not working, we could run `pre-commit` [post merge into main](https://docs.github.com/en/actions/writing-workflows/choosing-when-your-workflow-runs/events-that-trigger-workflows#running-your-pull_request-workflow-when-a-pull-request-merges) which would at least let us know we'd let something slip in to main but obviously not ideal.

Shehrozkashif, 2025-05-17:
Thanks for the detailed explanation, that really helps clarify things!

So just to confirm  if we can ensure that commit squashing happens *before* the PR enters the merge queue, then we should be able to use the `merge_group` event to run `pre-commit run commitlint`. But the challenge is that during the merge queue, `pre-commit` might not detect any changes because the correct `--from-ref` and `--to-ref` aren’t clearly defined?

If that’s the main blocker, would it be possible to explicitly set those refs in the action using something like:

```bash
pre-commit run commitlint --from-ref origin/main --to-ref HEAD
```

Or is there a risk that this won’t reflect the final squash commit accurately?

Also, is there any documentation or prior art you’ve seen where someone successfully validated squash commit messages during merge queue workflows? That might help guide the implementation here.

kbroch-rivosinc, 2025-05-17:
I don't think comparing from `origin/main` to `HEAD` works because if there are multiple PRs in the merge queue and say PR1 fails for commitlint violation then PR2 will fail for the same reason for the commit msg in PR1. 
Instead need to go from a github variable: https://docs.github.com/en/actions/writing-workflows/choosing-what-your-workflow-does/store-information-in-variables#default-environment-variables

Doc's for pre-commit https://docs.github.com/en/actions/writing-workflows/choosing-what-your-workflow-does/store-information-in-variables#default-environment-variables
but I don't think they account for complexity of merge queue (or squash merges).  I don't have prior art on a working example.  Perhaps you can search GH for something.  Otherwise I think we can just find hopefully the correct GH default enviro var and with some debug print statements in CI conclude we are doing the correct thing?

Shehrozkashif, 2025-05-18:
Got it, thanks for the detailed explanation! I’ll explore the GitHub env vars and add some debug prints to see what we can reliably use in the merge queue. I’ll also check for any existing solutions in public repos. Appreciate the guidance!