riscv/riscv-unified-db #697: Add date & time stamp to Python output to match Ruby

james-ball-qualcomm, 2025-04-29:
Add date & time stamp to Python output. See PR #668 for recommended Python code. See issue #664.

ThinkOpenly, 2025-04-30:
I posted some example code in the PR that James referenced above (#668).

AFOliveira, 2025-04-30:
> I posted some example code in the PR that James referenced above ([#668](https://github.com/riscv-software-src/riscv-unified-db/pull/668)).

Oh, my bad, that was "hidden", is that something you want to do then?

ThinkOpenly, 2025-04-30:
> > I posted some example code in the PR that James referenced above ([#668](https://github.com/riscv-software-src/riscv-unified-db/pull/668)).
> 
> Oh, my bad, that was "hidden", is that something you want to do then?

Either way. I may not get to it for a while. This issue is up for grabs unless/until then.

kbroch-rivosinc, 2025-04-30:
> [@kbroch-rivosinc](https://github.com/kbroch-rivosinc) do you know any good way of making this? If not, I can look it up

python standard logging is probably good enough.  Of course with python you have options: https://betterstack.com/community/guides/logging/best-python-logging-libraries/#1-the-standard-logging-module

From that list I've used standard and loguru.  I really liked loguru (it uses standard logging under the hood) but has nice defaults w/o much coding.

AFOliveira, 2025-05-02:
After giving this a second look, there is no need for python logging.

All of the UDB scripts are run via `do` which subsequently uses the main `Rakefile`. So, for all top-level tasks, we can just use Ruby logging for it. In fact, @james-ball-qualcomm already created logging all the python code that is in UDB, look at this example https://github.com/riscv-software-src/riscv-unified-db/blob/1ef0ccabd53042c87769957e150d4969f83302b3/Rakefile#L507

The only reason we would want to to log in python would be to add more granularity to the logging, but TBF I don't think there is any need for it now. If you guys agree, I think we may just close this issue.

james-ball-qualcomm, 2025-05-05:
I agree it isn't required.

closed 2025-05-05 by james-ball-qualcomm