riscv/riscv-unified-db #704: Add "recommended" values for extension parameters

ThinkOpenly, 2025-04-30:
**Is your feature request related to a problem? Please describe.**
Currently, extension parameters are defined with types and possible values ("Sdtrig" extension, Machine Context (`mcontext`), "hcontext" field; from PR #524):
```
params:
  DBG_HCONTEXT_WIDTH:
    schema:
      type: integer
      minimum: 0
      maximum: 14
    description: Specifies the size of HCONTEXT
```
Some parameters have "recommended" values in the spec, for example (Machine Context, `mcontext`, "hcontext" field):
> An implementation may tie any number of upper bits in this field to 0. If the H extension is not implemented, it’s recommended to implement 6 bits on RV32 and 13 bits on RV64 (as visible through the mcontext register). If the H extension is implemented, it’s recommended to implement 7 bits on RV32 and 14 bits on RV64.

**Describe the solution you'd like**
Some means of expressing a recommended value for each type (integer, boolean, etc.), including with the conditional logic required by the example above, seems desirable to allow configurations to not have to determine the recommended values and explicitly set all of the configuration parameters to those values.

ThinkOpenly, 2025-07-27:
> [@ThinkOpenly](https://github.com/ThinkOpenly) I explored this issue and wanted to confirm if the following approach aligns with your expectations:
> 
> * Add a `recommended` field to the `param_data` definition in `spec/schemas/ext_schema.json`, supporting:
>   
>   * Static values (`integer`, `string`, `boolean`, etc.)
>   * Conditional objects (e.g., based on architecture width or extension presence)

We may want to see if the "default" attribute can be made to work here, rather than a custom "recommended" attribute. This is some subtlety between the linguistic definitions of "default" and "recommended", but since it can be easily overwritten in a config file, it may not matter.

> Example implementation:
> 
> * Update a real parameter in `spec/std/isa/ext/Sdtrig.yaml` (e.g., `MSTATEEN_CONTEXT_TYPE` or `HSTATEEN_CONTEXT_TYPE`)
> * Include both:
>   
>   * A simple static `recommended` value
>   * A conditional `recommended` example (with inline comments for clarity)

Interesting choice. What do you propose as the default value for these parameters?

xenonnn4w, 2025-07-27:
Thanks! 
> We may want to see if the "default" attribute can be made to work here, rather than a custom "recommended" attribute. This is some subtlety between the linguistic definitions of "default" and "recommended", but since it can be easily overwritten in a config file, it may not matter.

i see, The `ext_schema.json` already supports JSON Schema defaults, so it will be better option as also mentioned by the Derek.

> Interesting choice. What do you propose as the default value for these parameters?

For the Sdtrig extension parameters, I propose `"rw"` as the default value for both `MSTATEEN_CONTEXT_TYPE` and `HSTATEEN_CONTEXT_TYPE`. It provides most flexibility by allowing read-write access to the CONTEXT bits, which is the least restrictive option that enables full functionality while still allowing implementations to override to more restrictive values (`read-only-0` or `read-only-1`) if needed. The `"rw"` default follows the principle of least surprise and maximizes compatibility, as it doesn't artificially limit access that could be useful for debugging or configuration purposes.