riscv/riscv-unified-db #705: Errata noticed in PR #638 (LLVM assembler alignment, vector syntax updates)

dbrash, 2025-04-30:
Looking at the output from the [#638](https://github.com/riscv-software-src/riscv-unified-db/pull/638) merge, I've noticed a couple of errata remain in the syntax output wrt 'x', 'f', and 'v' register nomenclature:

   - The vsetivli instruction update has changed xd to rd?
   - 39 instances of "rs1" associated with vector-float instructions should be fs1?
      - 35x instructions with the .vf suffix
      - 2x vfmv{.s.f, .v.f} instructions
      - vfwsub.wf and vfwsub.wf instructions
   - The vfmv.f.s instruction was 'vs2, xd', is now 'vd, vs2' and should be 'fd, vs2'?

David.

AFOliveira, 2025-04-30:
Thanks for the errata @dbrash! 
I am aware of that, I cut some corners on that PR to make sure I was fast enough to merge that assembly info before the summit. It was easier to identify "x" registers than to identify "f" registers on LLVM tablegen and since I believe the order of operands is a order of magnitude more important than type, I only identified the "easier" ones and delayed the FP registers to the future. 
I must admit I forgot to do the post-fix,  so your issue appeared at a nice timing. I will do this at my earliest availability now.

AFOliveira, 2025-05-07:
I am working on this now, and I realized that I was not aware of that last point:

> The vfmv.f.s instruction was 'vs2, xd', is now 'vd, vs2' and should be 'fd, vs2'?

Indeed, for all I can see LLVM has a bug on this and is describing this instruction wrongly. @lenary Do you think you can confirm this, based on the following snippet:

```
  "VFMV_F_S": {
    "!anonymous": false,
    "!fields": [
      "Inst",
      "SoftFail"
    ],
    "!locs": [
      "RISCVInstrInfoV.td:1669"
    ],
    "!name": "VFMV_F_S",
    "!superclasses": [
      "InstructionEncoding",
      "Instruction",
      "RVInstCommon",
      "RVInst",
      "RVInstV",
      "Sched"
    ],
    "AddedComplexity": 0,
    "AsmMatchConverter": "",
    "AsmString": "vfmv.f.s\t$vd, $vs2",
    "AsmVariantName": "",
    "CodeSize": 0,
    "Constraints": "",
    "DecoderMethod": "",
    "DecoderNamespace": "",
    "Defs": [],
    "DestEEW": {
      "def": "EEWSEWx1",
      "kind": "def",
      "printable": "EEWSEWx1"
    },
    "DisableEncoding": "",
    "ElementsDependOn": {
      "def": "EltDepsNone",
      "kind": "def",
      "printable": "EltDepsNone"
    },
    "EncodingInfos": null,
    "FastISelShouldIgnore": 0,
    "ForceTailAgnostic": 0,
    "HasPositionOrder": 0,
    "HasRoundModeOp": 0,
    "HasSEWOp": 0,
    "HasVLOp": 0,
    "HasVecPolicyOp": 0,
    "InOperandList": {
      "args": [
        [
          {
            "def": "VR",
            "kind": "def",
            "printable": "VR"
          },
          "vs2"
        ]
      ],
      "kind": "dag",
      "operator": {
        "def": "ins",
        "kind": "def",
        "printable": "ins"
      },
[... 305 lines cut ...]
    "isTrap": 0,
    "isUnpredicable": 0,
    "mayLoad": 0,
    "mayRaiseFPException": 0,
    "mayStore": 0,
    "usesCustomInserter": 0,
    "variadicOpsAreDefs": 0,
    "vd": [
      null,
      null,
      null,
      null,
      null
    ],
    "vm": 1,
    "vs2": [
      null,
      null,
      null,
      null,
      null
    ]
  },

```

AFOliveira, 2025-05-07:
Also, @dbrash I was checking the output of what you mentioned and could not identify that error in the ASM syntax, rather there still are some registers are still not correctly identified as FPR in the encoding section, was that what you were mentioning?

ThinkOpenly, 2025-05-07:
```
    "AsmString": "vfmv.f.s\t$vd, $vs2",
[...]
    "OutOperandList": {
      "args": [
        [
          {
            "def": "FPR32",
            "kind": "def",
            "printable": "FPR32"
          },
          "vd"
        ]
      ],
[...]
      "printable": "(outs FPR32:$vd)"
    },

```

"vd" is "FPR32". So, I think it's just that the name is arguably confusing or "wrong" for some definition of wrong. It likely "works". Looking at other LLVM files, they appear to use declarative typing ("FPR32:$rs2"), so the name is just a name.

It should probably be changed in LLVM, though, to reduce confusion.

It should be changed here as well.

lenary, 2025-05-08:
> Indeed, for all I can see LLVM has a bug on this and is describing this instruction wrongly. [@lenary](https://github.com/lenary) Do you think you can confirm this, based on the following snippet:

LLVM does not expose the operand names to its interfaces, and they are considered internal (even though they are exposed through tablegen).

How LLVM models its instructions is complex (we end up with two operands when an instruction updates a register, one to model the incoming value, and one to model the outgoing value, commonly named with `_wb` to stand for "writeback"). Sometimes these are not modelled with ins/outs, e.g. when the instruction reads/writes a specific register not in the encoding, when they're modelled with uses/defs instead.

> It should probably be changed in LLVM, though, to reduce confusion.

I could in this case, but I'm not sure we can in general indefinitely.

I think the LLVM matching code probably needs to create a mapping between the llvm names and the unified db operand names, rather than assuming that the names will always be the same.

AFOliveira, 2025-05-08:
maybe the most straightforward way would be to do is then associating any "d" register with its type and generate the name from there as @ThinkOpenly suggested. Thank you so much for the clarification @lenary!

AFOliveira, 2025-05-10:
Hi @dbrash, can you please clarify this? I could not find them. The rest are either solved or have PRs open to solve them.

>  39 instances of "rs1" associated with vector-float instructions should be fs1?
> 35x instructions with the .vf suffix
> 2x vfmv{.s.f, .v.f} instructions
> vfwsub.wf and vfwsub.wf instructions
> 

If you may confirm on this PDF, I would really appreciate it 

[instructions_appendix.pdf](https://github.com/user-attachments/files/20138722/instructions_appendix.pdf)

ThinkOpenly, 2025-05-14:
Is this related, or at least adjacent?
```
$ grep -e '- name: rs1' $(find arch/inst/V -name '*.yaml') | wc -l
428
$ grep -e '- name: rs1' $(find arch/inst/ -name '*.yaml' | grep -v arch/inst/V ) | wc -l
309
```
(Lots of uses of "rs1" still in the DB.)

AFOliveira, 2025-05-18:
@ThinkOpenly those certainly weren't introduced by my PR but can be fixed by it. Maybe a different issue, so that we can correctly categorize it would be good. I'm just waiting for David's approval to close this since I think they are fixed by now. The new issue maybe could be something along the lines of "Remove legacy rs1"

dbrash, 2025-05-19:
Alfonso,

Apologies for the delay - I wanted to get back from the Summit and have access to a monitor to check this out. I used your list of updates in issue #638 for my original checks. The instructions_appendix.pdf file you posted looks correct for the "fs1" updates, having caught all the cases I mentioned*. It also now has the vfwmaccbf16.vf case entry Zfbfwma too.

*I did find a couple of errors in my. 39-instance summary which may have confused you:
1. .wf variants should have been vfwadd and vfwsub (not 2x vfwsub)
2. I'd listed vfmerge.vfm as a .vf instruction

Otherwise:

The wavedrom diagrams still use the rd, rs1, rs2 formatting in many cases (not all) eg. add versus add.uw.

The vsetivli instruction update now  indicates xd correctly in the mnemonic (rd in the encoding data).

The vfmv.f.s instruction is still showing as 'vd, vs2'; should be 'fd, vs2'? It is rd in the encoding information.

commit 6ecca9c5 (2025-05-28): fix(encodings): fix V extension encodings mismatch with assembly (#812)

Fixes #705. I believe the main problem @dbrash comment in
https://github.com/riscv-software-src/riscv-unified-db/issues/705#issuecomment-2891042479
was that encodings and assembly were not matching, since I had fixed
assembly but not encodings. This PR should fix that.

---------
closed 2025-05-28 by AFOliveira