riscv/riscv-unified-db #707: test:regress bogged down by regress-gen-instruction-appendix

james-ball-qualcomm, 2025-04-30:
The 2nd longest running regression test task is 7 minutes. Can we see about speeding up this one:

- regress-gen-instruction-appendix

Does it need to be in the regression tests?

[image]

dhower-qc, 2025-04-30:
It spends the majority of its time running asciidoctor-pdf. I don't think we are going to get that sped up.

And, yes, it's important that this regression stays.

dhower-qc, 2025-04-30:
Digging a bit deeper, asciidoctor-pdf spends a lot of its time waiting for wavedrom-cli to generate all the register diagrams.

AFOliveira, 2025-05-02:
Maybe we could generate only a subset of this? The thing is that was not possible as it was before, but maybe the new method in #708 allows for us to config a smaller sub-set? This likely needs a feature addition to  instruction appendix generationand I may have to dig the new method to make it work

dhower-qc, 2025-05-05:
We could also just stop at adoc generation for the regression test.

AFOliveira, 2025-05-05:
> We could also just stop at adoc generation for the regression test.

Maybe we could stop at adoc just for this one. This way, the adoc build gets tested for all the other outputs, which should mean that nothing is broken either way, right?